WildernessLabs/Meadow_Issues #495: MicroGraphics - DrawVerticalGradient has a rendering glitch

jorgedevs, 2024-03-02:
The following code draws a vertical gradient:

```csharp
graphics.DrawVerticalGradient(
    x: 20,
    y: 20,
    width: graphics.Width - 40,
    height: graphics.Height - 40,
    colorTop: Color.Red,
    colorBottom: Color.Green);
```

Result:

[image]

Shouldn't it be a smooth gradient across the entire width and height?

adrianstevens, 2024-03-03:
The gradient is on the wrong orientation too .... can you share your project?

adrianstevens, 2024-03-03:
I can't repro here - definitely need to see your setup

jorgedevs, 2024-03-03:
For viz, the issue was found and solved.

adrianstevens, 2024-03-07:
shipped - closing issue

closed 2024-03-07 by adrianstevens