WildernessLabs/Meadow_Issues #496: CLI V2 Attempts to access invalid port

NevynUK, 2024-03-04:
CLI V 2.0.19

Hardware set up:

So I have some debug boards connected through a couple of hubs and I have these connected to a PL3.  The setup has the following port settings:

* Meadow: /dev/tty.usbmodem337B317434381
* ESP-PROG: /dev/tty.usbserial-1111101 and /dev/tty.usbserial-1111100
* ST-Link serial (to COM1 on Meadow): /dev/tty.usbmodem1111303

Attempting to flash firmware and the following commands have been executed:

* usepl (this is the alias _meadow config route /dev/tty.usbmodem337B317434381_)
* meadow device info
* meadow firmware write

Device info confirms that the Meadow is up and running and shows the following:

```
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: WiFiBasics

Hardware Information
    Processor type: STM32F777IIK6
    ID: 35-00-43-00-07-51-38-34-35-31-38-33
    Serial number: 337B31743438
    Coprocessor type: ESP32
    MAC Address -
	WiFi: 4C:75:25:D5:28:60

Firmware Versions
    OS: 1.9.0.0
    Runtime: 1.9.1.1
    Coprocessor: 1.9.1.0
    Protocol: 8
```

The `firmware write` is getting confused, output is:

```
Writing all firmware for version '1.9.0.0'
DFU Device Detected - using DFU to write OS
Flashing OS with /Users/markstevens/Library/Application Support/WildernessLabs/Firmware/1.9.0.0/Meadow.OS.bin
dfu-util 0.11

********** Some DFU output cut **********

Download done.	[=========================] 100%      2097152 bytes
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
[null]
Meadow found at /dev/tty.usbmodem1111303
No device found: Access to the port '/dev/tty.usbmodem1111303' is denied.
```

So despite being told which serial port to use it is trying to access the ST-Link which it cannot do as it has a terminal application connected to it.  So two issues really:

* Why is it trying to connect via the ST-Link?
* How can it find the Meadow device when it cannot actually connect to the terminal port?

Note that if I disconnect the terminal app then CLI finds the Meadow board correctly.

adrianstevens, 2024-03-04:
CLI v2 looks for a new serial port after a DFU flash which makes sense in most cases because the Meadow doesn't create a port in DFU mode but does after the reset.

And flashing via DFU doesn't use the serial port route because the serial port name after a flash is generally non-deterministic.

So it tries to be clever but clearly isn't working in this case. I have a few thoughts on how we can approach this and make it more robust in general.

NevynUK, 2024-03-04:
It is these two messages that seem odd:

```
Meadow found at /dev/tty.usbmodem1111303
No device found: Access to the port '/dev/tty.usbmodem1111303' is denied.
```

The two messages feel inconsistent.  The terminal app was connected to this port (/dev/tty.usbmodem1111303) all through the command sequence.

So, `Meadow found at /dev/tty.usbmodem1111303` should not have succeeded as the port should be locked by the terminal application as is indicated by the second message that the port is locked.

adrianstevens, 2024-03-07:
This should be fixed in v2.0.20.0

closed 2024-03-19 by adrianstevens