WildernessLabs/Meadow_Issues #497: Unhandled Exceptions thrown in a background Task do not bubble up

ctacke, 2024-03-04:
This "unhandled" exception is not caught or handled by MeadowOS.  I'd expect it to end up in either MeadowOS.cs line 54 or line 332 but the app just appears to die quietly

```
public override Task Run()
{
    StartHeartbeat();

    return Task.CompletedTask;
}

protected void StartHeartbeat()
{
    Task.Run(async () =>
    {
            throw new Exception("Testing restart...");
    });
}
```

doingnz, 2024-03-05:
Here is an example application you may want to try
```
using Meadow;
using Meadow.Devices;
using System;
using System.Threading;
using System.Threading.Tasks;
using static System.Net.Mime.MediaTypeNames;

namespace deleteme
{
    public class MeadowApp : App<F7CoreComputeV2>
    {
        public override Task Initialize()
        {
            Resolver.Log.Info("Initialize...");

            Resolver.Log.Info("Uses Resolver.Log");

            Task.Run(async () =>
            {
                while (true)
                {
                    Resolver.Log.Info("Blink...");
                    await Task.Delay(100);
                }
            });

            AppDomain.CurrentDomain.UnhandledException += new UnhandledExceptionEventHandler(CurrentDomain_UnhandledException);

            return Task.CompletedTask;
        }

        private void CurrentDomain_UnhandledException(object sender, UnhandledExceptionEventArgs e)
        {
            if (e.ExceptionObject is Exception exception)
            {
                Resolver.Log.Info($"CurrentDomain_UnhandledException: {exception.Message}");
            }
            else
            {
                Resolver.Log.Info($"CurrentDomain_UnhandledException: {e}");
            }
        }

        public override async Task Run()
        {
            Resolver.Log.Info("Run...");

            Resolver.Log.Info("Hello, Meadow Core-Compute!");

            StartHeartbeat();

            while (!badWorker.IsCompleted) { Console.Write("."); await Task.Delay(2000); }

            if (badWorker.Status == TaskStatus.Faulted)
            {
                foreach (var e in badWorker.Exception.InnerExceptions)
                {
                    // Handle the custom exception.
                    //if (e is CustomException)
                    //{
                    //    Console.WriteLine(e.Message);
                    //}
                    //// Rethrow any other exception.
                    //else
                    {
                        throw e;
                    }
                }
            }

            //Task.Delay(5000);

            Resolver.Log.Info("Run Task.CompletedTask");

            return;// Task.CompletedTask;
        }

        private static Task? badWorker;

        protected void StartHeartbeat()
        {
            badWorker = Task.Run(async () =>
            {
                Resolver.Log.Info("StartHeartbeat...");
                throw new Exception("Testing restart...");
            });
        }
    }
}
```

doingnz, 2024-03-05:
running the above application produces the following output on a CCM running 1.9.0.0 Meadow OS
```
Initializing App
Initialize...
Uses Resolver.Log
Running App
Run...
Hello, Meadow Core-Compute!
.
Blink...
StartHeartbeat...
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
App Error in App Run: Testing restart...
System.Exception: Testing restart...
Blink...
Blink...
at deleteme.MeadowApp.Run () [0x00108] in C:\Users\centr\source\repos\deleteme\MeadowApp.cs:66
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0c00f68 + 0x00236> in <79b339c57f5b49268b3d59b838a2fcec>:0
App shutting down
Shutdown
Blink...
Blink...
Blink...
Blink...
```

doingnz, 2024-03-05:
Notes
1. The exception info is available if you have access to the Task object that threw the exception (here it's badWorker). 
2. It is possible to set breakpoints in the app and step them prior to the fatal throw e;
3. UnhandledExceptionEventHandler(CurrentDomain_UnhandledException); is not called (should be called before the throw e; on the "Run" Task?)
4. The App goes into Shutdown , however the "Blink..." continues forever.

ctacke, 2024-03-06:
Question for you, @doingnz .  I see this in the output:
```
Shutdown
Blink...
Blink...
Blink...
```
Do you have the device configured to restart on app failure?

`app.config.yaml`
```
Lifecycle:
  ResetOnAppFailure: true
```

This should schedule a device restart via the watchdog, even if other threads are running.

doingnz, 2024-03-06:
In the previous test app.config.yaml was not set to restart automatically.

If I change to `RestartOnAppFailure: true', then yes, Meadow will restart :-)

I assume there would be mixed value registering for a shutdown event notification and try clean up resources in use by a real world application... ability to clean up would depend what code crashed and what the app was doing.... :-)

```
StartHeartbeat...
Blink...
Blink...
App Error in App Run: Testing restart...
System.Exception: Testing restart...
Blink...
Blink...
Blink...
at deleteme.MeadowApp.Run () [0x00108] in <1a6742161a754ad993dbc89d2d755ba4>:0
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00176] in <79b339c57f5b49268b3d59b838a2fcec>:0
App shutting down
CRASH: Meadow will restart in 1 seconds.
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
Blink...
Resetting device...
```

alexischr, 2024-03-06:
I think that indeed we need to harden Meadow.Core so that transient network failures are recoverable. 

At the same time, non-network-related failures to start due to misconfiguration need to not reboot forever. An attempt to receive an corrective update should always happen.

ctacke, 2024-03-07:
marking as fixed.  It's replaced by two things: an audit of Core, and implementing crash reporting.

doingnz, 2024-03-07:
Does the https://github.com/WildernessLabs/Meadow_Issues/issues/520 CRASH reporting  fix the issue of not raising an unhandled exception event?

I should have highlighted the trace above when running the `deleteme` app does not include any indication the method `CurrentDomain_UnhandledException()` was called. 

Is this event supported by the F7 version of .Net?

```
AppDomain.CurrentDomain.UnhandledException += new UnhandledExceptionEventHandler(CurrentDomain_UnhandledException);
```

closed 2024-03-26 by ctacke

ctacke, 2024-03-26:
So the event is supported, but we're consuming it in Core, and it appears that it does not call it twice.  Our handlers does, however, call out to the `App` when it occurs, so if you add either of these, you will get notified when the app is going down:
```
public override Task OnShutdown()
{
    Resolver.Log.Info($"OnShutdown called");

    return base.OnShutdown();
}

public override Task OnError(Exception e)
{
    Resolver.Log.Info($"OnError called with {e.Message}");
    return base.OnError(e);
}
```

doingnz, 2024-03-27:
@ctacke does call to event handler `public override Task OnError(Exception e)` imply a fatal error and the app will always be shut down after this has been called?