WildernessLabs/Meadow_Issues #501: CLI V2 `meadow firmware write -v 1.9.1.6` is not moving on to flashing the RT or ESP

NevynUK, 2024-03-05:
CLI V2.0.19
Mac OS

I am trying to get a new version of the OS and RT on the board.  The version number is the same as the ones already deployed as it is from a local development version of the code.

I have created a new firmware directory 1.9.1.6 which contains the binaries I need to write to the board and I run the command `meadow firmware write -v 1.9.1.6`.

CLI output is below:

```
Writing all firmware for version '1.9.1.6'
DFU Device Detected - using DFU to write OS
Flashing OS with /Users/markstevens/Library/Application Support/WildernessLabs/Firmware/1.9.1.6/Meadow.OS.bin
dfu-util 0.11

********** DFU Output Cut **********

Download done.	[=========================] 100%      2097152 bytes
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
[null]
Meadow found at /dev/tty.usbmodem337B317434381
Runtime is enabled
Initializing OS...

Parsing app.config.yaml...
```

So CLI has found the right serial port but it has not disabled mono and started the RT update process.  Instead the application on the board has started.

adrianstevens, 2024-03-06:
Looks like the OS behavior has changed - v1.9.0.0 works, v1.9.1.7 isn't providing the runtime state

NevynUK, 2024-03-06:
Change requested through OS PR [567](https://github.com/WildernessLabs/Meadow/pull/567)

Change amended in OS PR [570](https://github.com/WildernessLabs/Meadow/pull/570)

adrianstevens, 2024-03-07:
Also fixed in v2.0.20.0 of the CLI

closed 2024-03-07 by adrianstevens