WildernessLabs/Meadow_Issues #502: CLI V2 Flash failure resets route to device

NevynUK, 2024-03-05:
CLI V2.0.19
MacOS

So fixing #496 may resolve this

Starting conditions:

* Meadow board with 1.9.0.0 OS and RT on the board.
* Project Lab board with debug connectors attached to ESP and STM debug ports
* CoolTerm application connected to the ST-Link serial port

Command executed: `meadow firmware write -v 1.9.1.6`

I get the same output as discussed in #496 where the final CLI output is:

```
Meadow found at /dev/tty.usbmodem1111303
No device found: Access to the port '/dev/tty.usbmodem1111303' is denied.
```

I then reset the board and try `meadow device info` and I receive the output:

```
Getting device info...
No device found: Access to the port '/dev/tty.usbmodem1111303' is denied.
```

So CLI has decided to reset the route to the board I am using.  I have previously configured the route to the device and it is overriding my wishes - how rude.

adrianstevens, 2024-03-06:
ahhh yes it definitely will - I'll fix

adrianstevens, 2024-03-07:
should be fixed in v2.0.20.0 but not fully tested

closed 2024-03-19 by adrianstevens