WildernessLabs/Meadow_Issues #506: Index out of range exceptions in Esp32Coprocessor.Encoder.cs on RaiseWiFiDisconnected

doingnz, 2024-03-05:
When the Wifi disconnects, events are raised that will call `Esp32WiFiAdapter.RaiseWiFiDisconnected(...)` with an assumed to be valid `byte[] payload` parameter. In turn this calls `Encoders.ExtractDisconnectEventData(payload, 0);` to decode the reason for the disconnect. 

Encoders.ExtractDisconnectEventData(byte[] buffer, int offset) does not check it is called with valid payload data and simply indexes into the byte array in a for...loop. `(buffer[index + offset] != 0)`. etc.

This will cause `System.IndexOutOfRangeException: Index was outside the bounds of the array.` if for example payload has zero bytes length. 

Issue WildernessLabs/Meadow_Issues#497 means the application simple stops execution without any error messages. 

Should all the methods in Encoders.cs at a minimum have try/catch clauses? Simple sanity checks would also fail faster than try catch.
```
            if (buffer.Length == 0)
            {
                return (disconnectEventData); 
            }
```

ctacke, 2024-03-07:
The OS team is investigating this.  The fix feels right, but there's a question of why bad data is even getting in here that is being chased.

doingnz, 2024-03-14:
Currently Meadow will crash / shutdown any time wifi is turned off rather than the expected behaviour of being notified by an event.  Personally I think this is a p0 bug. 

I appreciate the team wants to understand why `RaiseWiFiDisconnected` is called with a zero length payload[] and given the method `ExtractDisconnectEventData` is auto generated, it appears the way forward, even if only till some other solution is found, is to apply some very simple sanity checks in `RaiseWiFiDisconnected` as follows. 

Let me know if you want me to make a new PR with this change?

```
    protected void RaiseWiFiDisconnected(StatusCodes statusCode, byte[] payload)
    {
        ClearNetworkDetails();
        DisconnectEventData disconnectEventData = new DisconnectEventData
        {
            Reason = (byte)WiFiDisconnectedReason.Unspecified
        };
        if (payload.Length != 0)
        {
            try
            {
                disconnectEventData = Encoders.ExtractDisconnectEventData(payload, 0);
            }
            catch (Exception e)
            {
                // ignore exception and use default "WiFiDisconnectedReason.Unspecified"
            }
        }

        string reason = DisconnectReason(disconnectEventData);
        lock (_lock)
        {
            RaiseNetworkDisconnected(new NetworkDisconnectionEventArgs(reason));
        }
    }
```

adrianstevens, 2024-08-20:
@NevynUK any progress on this one?

halyssonJr, 2025-01-02:
This issue was fixed. I tested in two scenarios:
1) Disconnect manually
2) Turn off the router (force disconnection)

### Result 1
```
wifi.IsConnected = False
IPAddress = 255.255.255.255
Waiting for wifi to connect loop
wifi.IsConnected = True
IPAddress = 255.255.255.255
Waiting for wifi to connect loop
wifi.IsConnected = True
IPAddress = 255.255.255.255
Waiting for wifi to connect loop ended
wifi.IsConnected = True
IPAddress = 192.168.15.2
Wifi_NetworkConnected event fired...
wifi.IsConnected = True
Execute wifi.Disconnect now
wifi.IsConnected (from Wifi_NetworkDisconnected event) = False
Reason - ManualDisconnect
wifi.IsConnected = False
IPAddress = 255.255.255.255
```

### Result 2
```
IPAddress = 255.255.255.255
Waiting for wifi to connect loop ended
wifi.IsConnected = True
IPAddress = 192.168.15.2
Wifi_NetworkConnected event fired...
wifi.IsConnected = True
wifi.IsConnected (from Wifi_NetworkDisconnected event) = True
Reason - 0
Execute wifi.Disconnect now
wifi.IsConnected = False
IPAddress = 255.255.255.255
Wifi_NetworkConnected event fired...
wifi.IsConnected = True
```

halyssonJr, 2025-01-02:
@doingnz, could please check those results and see if you agree?

alexischr, 2025-01-07:
Closing for now as we have not seen this behavior in a while and after re-testing, and this was almost certainly a side-effect of resolved issue https://github.com/WildernessLabs/Meadow_Issues/issues/765

closed 2025-01-07 by alexischr