WildernessLabs/Meadow_Issues #507: Memory leak in MeadowCloudService

DarkNoir, 2024-03-05:
MeadowCloudService needs to dispose of HttpResponseMessage and StringContent
@bryancostanich

duduita, 2024-05-18:
It seems that we have a memory leak in this [Send](https://github.com/WildernessLabs/Meadow.Core/blob/6d47d040979d4533047518b7678e75bb10cba6dc/source/Meadow.Core/Cloud/MeadowCloudConnectionService.cs#L698) method.

Hopefully, we have an easy way to reproduce, we just need to throw an artificial exception after this [Post](https://github.com/WildernessLabs/Meadow.Core/blob/059824d6d56697c397a4480c21333ddf1edc87a8/source/Meadow.Core/Cloud/MeadowCloudConnectionService.cs#L734), and wait 30 minutes ~ 1 hour, until we see the OOM error.

alexischr, 2024-06-11:
We have added eager disposal of `HttpResponseMessage` et al. in MeadowCloudService

closed 2024-06-18 by bryancostanich