WildernessLabs/Meadow_Issues #516: Meadow CLIv2 - Initial Flash process not starting

jorgedevs, 2024-03-05:
On:

- Meadow.CLI 2.0.19
- Windows 11 (Latest version)
- Project Lab v3 in good state (STM32 BOOTLOADER/Serial COM Port in bootloader in regular mode)
- DFU installed

When trying to flash a device using `meadow firmware write` command, it outputs:

```console
PS C:\WINDOWS\system32> meadow firmware write
Writing all firmware for version '1.9.0.0'
DFU Device Detected - using DFU to write OS
Flashing OS with C:\Users\jk225\AppData\Local\WildernessLabs\Firmware\1.9.0.0\Meadow.OS.bin
dfu-util 0.11
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
Warning: Invalid DFU suffix signature
New meadow device not found
```

adrianstevens, 2024-03-06:
it's missing a lot of output 

let's try running `dfu-util --list` while the device is in dfu mode

And I'd like to see the command run from a command prompt instead of powershell (just to be sure)

j-koda80, 2024-03-08:
1) First screenshot is of my command prompt while in bootloader mode.
2) Second screenshot of device manager showing device state while running step 1.
3) Third screenshot is of my comment prompt while device in regular mode showing the device is found.
4) Fourth screenshot is of my device manager showing device state while running step 3.
[image]
[image]
[image]
[image]

adrianstevens, 2024-03-08:
Can you try flashing again with the latest version of the CLI? v2.0.22.0?

`dotnet tool update WildernessLabs.Meadow.CLI --global`

adrianstevens, 2024-03-12:
This _might_ be fixed with v2.0.23 .... otherwise we'll probably need to hop on a call and debug

j-koda80, 2024-03-14:
[image]

@adrianstevens, Still an issue unfortunately, let's hop on a call!

adrianstevens, 2024-03-15:
My current theory is the hack to fix Mark's machine broke Jora's :) 

I published a beta CLI - v2.0.26-beta that I _believe_ will fix the issue.

jorgedevs, 2024-03-15:
We tracked down the issue and it has been fixed.

closed 2024-03-19 by adrianstevens