WildernessLabs/Meadow_Issues #521: Driver Request: ADT7410 High Accuracy I2C Temperature Sensor

kingdona, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
No problems. Need integration with ADT7410 High Accuracy I2C Temperature Sensor

**Describe the solution you'd like**
I would like a driver to be able to use this sensor.

**Describe alternatives you've considered**
I have not tried other sensors for this application. It is part of a combo featherwing I would like to use for my project.
https://www.adafruit.com/product/4147

**Additional context**

https://www.adafruit.com/product/4089)
https://www.analog.com/media/en/technical-documentation/data-sheets/adt7410.pdf

adrianstevens, 2024-03-10:
The driver looks like it should be pretty straight forward https://github.com/adafruit/Adafruit_ADT7410/blob/master/Adafruit_ADT7410.cpp

I'll order some hardware and we get it built

adrianstevens, 2024-03-26:
Hardware is ordered - I'll update when it arrives

adrianstevens, 2024-05-03:
Drivers for the ATD7410 and ADXL343 have been added and will ship in the next release

closed 2024-05-28 by adrianstevens