WildernessLabs/Meadow_Issues #528: Meadow.CLI v2 - Getting a Semaphore timeout when flashing Feather v1s

jorgedevs, 2024-03-15:
Trying to flash a couple of F7 Feather v1 boards, im getting a semaphore timeout issue:

Using Meadow.CLI v2.24

```console
C:\Users\ramir>meadow firmware write --v 1.9.0.0
Writing all firmware for version '1.9.0.0'
DFU Device Detected - using DFU to write OS
Flashing OS with C:\Users\ramir\AppData\Local\WildernessLabs\Firmware\1.9.0.0\Meadow.OS.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 2097152

Erase    done.  [=========================] 100%      2097152 bytes

Download done.  [=========================] 100%      2097152 bytes
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
Warning: Invalid DFU suffix signature
Meadow found at COM17

Getting runtime for 1.9.0.0...

Writing runtime...
Writing Meadow.OS.Runtime.bin: 10%
Writing Meadow.OS.Runtime.bin: 20%
Writing Meadow.OS.Runtime.bin: 30%
Writing Meadow.OS.Runtime.bin: 40%
Writing Meadow.OS.Runtime.bin: 50%
Writing Meadow.OS.Runtime.bin: 60%
Writing Meadow.OS.Runtime.bin: 70%
Writing Meadow.OS.Runtime.bin: 80%
Writing Meadow.OS.Runtime.bin: 90%
Writing Meadow.OS.Runtime.bin: 100%

The semaphore timeout period has expired. : 'COM17'
```

It's 100% reproducible on my Windows 11 (up-to-date) desktop machine.

adrianstevens, 2024-03-15:
I can flash v1 feathers here - Windows 11, flashing from 1.9.0.0 to 1.9.1.8 

Definitely need to track this down

jorgedevs, 2024-03-15:
I'll close this issue for now, since I was able to flash v1s boards lately. I'll re-open if/when I have more info and a definite repro.

closed 2024-03-15 by jorgedevs