WildernessLabs/Meadow_Issues #529: nolink doesn't appear to be nolinking, or is not behaving as epected

ctacke, 2024-03-15:
I have an F7 app that uses MeadowCloud commands.  The command is simple:
```csharp
class MyCommand
{
    public int Value { get; set; }
}
```

The command **is defined in another assembly** - `MyApp.Core`

When I send the command to the device, the `Value` get deserialized to zero when I send valid JSON with a value.
```json
{
    "Value": 42
}
```
I then added `app.build.yaml` to the app, set it to CopyAlways.

This is the file content:
```
Deploy:
  nolink: [ MyApp.Core ]
```
The build output indicates that it is seen.  This is in the output:
```
7>'app.build.yaml' found!
```
However when the value is deserialized, the property is *still zero*.  The fact that my app can read the property to display the zero tells me the property is there, but I suspect the property setter is getting linked out.  I can confirm this with the following workaround:

In my app I define a property and set it.  The property is never used anywhere, but it is public and is set

class MyApp : App<MeadowF7>
{
    public MyCommand Command { get; set; }

    public MyApp()
    {
        Command = new()
        {
            Value = 42
        };
    }
}

Now the deserialization works.

adrianstevens, 2024-03-15:
repro'd

```
var lib = new BlinkyLib.Class1();

Console.WriteLine($"Id: {lib.Id}");

//find the ID property on lib via reflection and set it to 42
PropertyInfo idProperty = lib.GetType().GetProperty("Id", BindingFlags.Public | BindingFlags.Instance);
if (idProperty != null && idProperty.CanWrite)
{
    idProperty.SetValue(lib, 42, null);
}

// Verify the change
Console.WriteLine($"Id after using reflection to set to 42: {lib.Id}");
```

adrianstevens, 2024-03-15:
works in the current version of the CLI .... there was a typo above

Deploy:
  nolink: [ MyApp.Core ]
  
  should be 
  
Deploy:
  NoLink: [ MyApp.Core ]
  
closing issue

closed 2024-03-15 by adrianstevens