WildernessLabs/Meadow_Issues #534: USB Port change each device reset on Linux

halyssonJr, 2024-03-19:
# Summary 
For Linux users, we have some issues regarding the USB port and Meadow CLI, which were fixed in the previous version (v 1. x.x). So the problem is, every time the device was restarted a new USB port was created, usually switching to the next index port, e.g.:

 /dev/ttyACM0 -> Reseting -> /dev/ttyACM1

What's expected?
 Automatically find the new port and increase a little bit the timeout.

**Workaround**  :  wait for the on-bord LED to blink and press ` Ctrl+C ` and select the new USB port.

# Result
```
$ meadow port select
Found device on port:
 1: /dev/ttyACM0

Setting route=/dev/ttyACM0

$ meadow runtime disable
Disabling runtime...
The operation has timed out

$ meadow port select
Found device on port:
 1: /dev/ttyACM1

Setting route=/dev/ttyACM1

```

ctacke, 2024-04-02:
I wonder if the port is still held by the CLI after the reset and the OS is then incrementing the device index.  We might want to see if we can verify and remedy that before doing a device search on all ports.  The reason I suspect this is because with the Raspberry Pi at least, when I unplug and re-plug the device, it keeps coming back at the same port (`/dev/ttyACM0`)

halyssonJr, 2025-02-26:
Testing the Meadow.CLI version `v2.0.67`on Linux, we have consistent behavior and better experience. I will close this issue, I can see any problem with TTY ports.

closed 2025-02-26 by halyssonJr