WildernessLabs/Meadow_Issues #535: Command output hard to read on macOS light mode

patridge, 2024-03-19:
Version: 2.0.3
macOS: Sonoma 14.0 (not latest, which is 14.3.1)

At least on this new macOS system I'm just getting rolling, when I run the `meadow` commands, the output is heavily light gray on a white background.

[image]

CartBlanche, 2024-04-03:
@patridge Could you tell me how you set your terminal window colour to light?

CartBlanche, 2024-04-04:
@patridge could you please try the CI built CLI here - https://github.com/WildernessLabs/Meadow.CLI/actions/runs/8557841541

closed 2024-04-04 by adrianstevens