WildernessLabs/Meadow_Issues #539: Update service error reauthenticating

lamebrain, 2024-03-19:
After reporting health metrics successfully, an error occurred while reauthenticating:
```
stdout> making cloud log httprequest with json: {"measurements":{"health.cpu_temp_celsius":44.9560439560439,"health.memory_used":1154184,"health.disk_space_used":17586715,"info.os_version":"1.8.0.0","health.battery_percentage":0,"info.coprocessor_os_version":"1.8.0.0"},"description":"device.health","eventId":10,"timestamp":"2024-03-19T22:14:01.188Z"}
stdout> Unauthorized, re-authenticating
stdout> UnhandledException
stdout> sending Meadow.Cloud.CloudLog
stdout> making cloud log httprequest with json: {"timestamp":"2024-03-19T22:14:04.654Z","severity":"Error","message":"UnhandledException","exception":null}
stdout>  System.InvalidOperationException: This instance has already started one or more requests. Properties can only be modified before sending the first request.
stdout>   at System.Net.Http.HttpClient.CheckDisposedOrStarted () [0x00010] in <d2e0ec4df24f4640966bc5a31bcd9656>:0 
  at System.Net.Http.HttpClient.set_BaseAddress (System.Uri value) [0x0000b] in <d2e0ec4df24f4640966bc5a31bcd9656>:0 
  at Meadow.MeadowCloudConnectionService.Send[T] (T item, System.String endpoint) [0x00078] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at Meadow.HealthRepor
stdout> er.Send () [0x002bf] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at Meadow.HealthReporter+<>c__DisplayClass1_0.<Start>b__0 (System.Object sender, System.Timers.ElapsedEventArgs e) [0x00079] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:
stdout>  
  at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e7054f2a5c394bb7bb05b4a0f
stdout> dd0bbb>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <e7054f2a5c3
stdout> 4bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
stdout> Shutdown
stdout> Cloud send failed. Reponse was ''
stdout> sending Meadow.Cloud.CloudLog
stdout> making cloud log httprequest with json: {"timestamp":"2024-03-19T22:14:04.833Z","severity":"Error","message":" System.InvalidOperationException: This instance has already started one or more requests. Properties can only be modified before sending the first request.","exception":null}
stdout> Cloud send failed. Reponse was ''
stdout> sending Meadow.Cloud.CloudLog
stdout> making cloud log httprequest with json: {"timestamp":"2024-03-19T22:14:05.64Z","severity":"Error","message":"  at System.Net.Http.HttpClient.CheckDisposedOrStarted () [0x00010] in <d2e0ec4df24f4640966bc5a31bcd9656>:0 
  at System.Net.Http.HttpClient.set_BaseAddress (System.Uri value) [0x0000b] in <d2e0ec4df24f4640966bc5a31bcd9656>:0 
  at Meadow.MeadowCloudConnectionService.Send[T
stdout>  (T item, System.String endpoint) [0x00078] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at Meadow.HealthReporter.Send () [0x002bf] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at Meadow.HealthReporter+<>c__DisplayClass1_0.<Start>b__0 (System.Object sender, System.Timers.ElapsedEventArgs e) [0x00079] in <fa7042d4d54f48179d5ad03205bf5181>:0 
  at System.Runtime.CompilerServices.AsyncMe
stdout> hodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCa
stdout> lback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threadin
stdout> .QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 ","exception":null}
stdout> Cloud send failed. Reponse was ''
stdout> sending Meadow.Cloud.CloudLog
stdout> making cloud log httprequest with json: {"timestamp":"2024-03-19T22:14:07.504Z","severity":"Warning","message":"Cloud send failed. Reponse was ''","exception":null}
stdout> Cloud send failed. Reponse was ''
stdout> sending Meadow.Cloud.CloudLog
stdout> making cloud log httprequest with json: {"timestamp":"2024-03-19T22:14:09.654Z","severity":"Warning","message":"Cloud send failed. Reponse was ''","exception":null}
stdout> Cloud send failed. Reponse was ''
```

ctacke, 2024-03-20:
If I change to using code to connect, it will connect (albeit slow AF - separate issue created for that). With auto-connect configured, I don't know how long it takes to connect.  Longer than I'm willing to wait.  Here's the current output for me:
```
Meadow.Cloud service waiting for network connection (420.555 s)
Meadow.Cloud service waiting for network connection (435.557 s)
Meadow.Cloud service waiting for network connection (450.559 s)
Meadow.Cloud service waiting for network connection (465.561 s)
Meadow.Cloud service waiting for network connection (480.563 s)
Meadow.Cloud service waiting for network connection (495.565 s)
Meadow.Cloud service waiting for network connection (510.567 s)
Meadow.Cloud service waiting for network connection (525.569 s)
Meadow.Cloud service waiting for network connection (540.571 s)
Meadow.Cloud service waiting for network connection (555.573 s)
Meadow.Cloud service waiting for network connection (570.575 s)
Meadow.Cloud service waiting for network connection (585.577 s)
Meadow.Cloud service waiting for network connection (600.579 s)
Meadow.Cloud service waiting for network connection (615.581 s)
```

ctacke, 2024-03-20:
Stranger note:  I changed access points and auto-connect worked after 60s.  Both networks are 2.4GHz.  One just never connects, and the ESP never gives an error.  I'll try adding at least a timeout error to the auto-connect logic

lamebrain, 2024-03-22:
With the fixes from https://github.com/WildernessLabs/Meadow_Issues/issues/537, I'm not longer getting this.

closed 2024-03-22 by lamebrain