WildernessLabs/Meadow_Issues #540: Add ability to list orgs from command line

ctacke, 2024-03-20:
I'm trying to provision my device and I get this error:

```
> meadow device provision
Retrieving your user and organization information...
You are a member of more than 1 organization. Please specify the desired orgId for this device provisioning.
```

I have no way to get a list of orgs from the CLI, so I have to go to the browser to look up the IDs.  It would be nice if the CLI has a way to list the orgs directly.

Maybe something like

```
meadow user list orgs
```

Maybe even listing them in the error message would be helpful as well.

adunndevster2, 2024-04-24:
Yeah, I second listing them in the error message, especially since we can only have a max of 3 today

adrianstevens, 2026-11-06:
Fixed in Meadow.CLI 3.0.3 (WildernessLabs/Meadow.CLI#669), on the meadow-3.0 branch.

closed 2026-11-06 by adrianstevens