WildernessLabs/Meadow_Issues #551: OS is not properly parsing `meadow.config.yaml`

ctacke, 2024-03-26:
When I configure the meadow using the following (which a linter says is valid) the network ends up using WiFi (0, which I assume is the default):
```
# Acceptable values for true: true, 1, yes
# Acceptable values for false: false, 0, no

# Main Device Config
Device:

    # Name of the device on the network.
    Name: MeadowDevice

    # Uncomment if SD card hardware present on this hardware 
    # (e.g., Core-Compute module with SD add-on)? Optional; default value is `false`.
#    SdStorageSupported: true

# Control how the ESP coprocessor will start and operate.
Coprocessor:

    # Should the ESP32 automatically attempt to connect to an access point at startup?
    # If set to true, wifi.yaml credentials must be stored in the device.
#    AutomaticallyStartNetwork: true

    # Should the ESP32 automatically reconnect to the configured access point?
#    AutomaticallyReconnect: true

    # Maximum number of retry attempts for connections etc. before an error code is returned.
#    MaximumRetryCount: 7

# Network configuration.
Network:

    # Network Interface (Default value is WiFi with DHCP)
    Interfaces:
        - Name: Ethernet
          UseDHCP: true
#        - Name: WiFi
#          UseDHCP: true
#          IPAddress:
#          NetMask:
#          Gateway:

    # Default Interface
    DefaultInterface: Ethernet

    # Automatically attempt to get the time at startup?
    GetNetworkTimeAtStartup: true

    # Time synchronization period in seconds.
    NtpRefreshPeriod: 600

    # Name of the NTP servers.
    NtpServers:
        - 0.pool.ntp.org
        - 1.pool.ntp.org
        - 2.pool.ntp.org
        - 3.pool.ntp.org

    # IP addresses of the DNS servers.
    DnsServers:
        - 1.1.1.1
        - 8.8.8.8
```
If I clean out some comments, etc and use this, it works as expected:
```
Device:
    # Name of the device on the network.
    Name: MeadowDevice

Network:
    Interfaces:
        - Name: Ethernet
          UseDHCP: true
    DefaultInterface: Ethernet

    # Automatically attempt to get the time at startup?
    GetNetworkTimeAtStartup: true

    # Time synchronization period in seconds.
    NtpRefreshPeriod: 600

    # Name of the NTP servers.
    NtpServers:
        - 0.pool.ntp.org
        - 1.pool.ntp.org
        - 2.pool.ntp.org
        - 3.pool.ntp.org

    # IP addresses of the DNS servers.
    DnsServers:
        - 1.1.1.1
        - 8.8.8.8
```

ctacke, 2024-03-26:
I suspect it's the empty `Coprocessor:` node.  Removing it allows things to work.

NevynUK, 2024-03-26:
It looks like any empty section will cause the issue.  So and empty `MonoControl` will also cause the issue.

NevynUK, 2024-03-26:
Log file contains:

```
I~2024-03-25T09:59:48Z~Logging started
E~2024-03-25T09:59:48Z~CONFIG: Load: Expecting MAPPING, got event: SCALAR
E~2024-03-25T09:59:49Z~CONFIG: Load: Backtrace:
E~2024-03-25T09:59:49Z~CONFIG:   in mapping field: MonoControl
E~2024-03-25T09:59:49Z~Error processing config file, using default config
I~2024-03-25T09:59:50Z~WiFi credentials file not found
```

halyssonJr, 2025-04-14:
@alexischr, sharing some result testing `OS 2.1.0.0`.

If you enable the debug option on `app.config.yaml`, it will show a message from the ESP adapter saying: " interface System, event code: OsException status code InvalidConfigurationFile (0x0000002b)".

```
Logging:

    # Adjust the level of logging detail.
    LogLevel:
    
        # Trace, Debug, Information, Warning, or Error
        Default: Trace
```
Console Output:
```
[esp] Starting Esp32Coprocessor event handler thread.
[esp] Waiting for event.
[esp] Event received.
[esp] Processing event.
[esp] Simple event, interface System, event code: OsException, status code InvalidConfigurationFile (0x0000002b)
[esp] Event data collected, raising event.
[esp] Waiting for event.
[core] Device is configured to use WiFi for the network interface
[core] The ESP32 has had an error (InvalidConfigurationFile).
```