WildernessLabs/Meadow_Issues #552: First Ethernet request is unacceptably slow

ctacke, 2024-03-26:
Using Core Compute breakout v2.b, OS 1.9

I am trying to do a Postman echo call using Ethernet using the Ethernaet sample app.  The first request takes just over 5 minutes to return, even if I explicitly set a timeout at 15 seconds.  Subsequent requests take about 1 second.

Same delay with `http` or `https` so I don't think it's SSL related.

Fundamentals of the sample app:

```
public override async Task Run()
{
    var ethernet = Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();
    await GetWebPageViaHttpClient("http://postman-echo.com/get?foo1=bar1&foo2=bar2")
}

public async Task GetWebPageViaHttpClient(string uri)
{
    Resolver.Log.Info($"Requesting {uri} - {DateTime.Now}");

    using (HttpClient client = new HttpClient())
    {
        client.Timeout = new TimeSpan(0, 0, 15);

        HttpResponseMessage response = await client.GetAsync(uri);

        try
        {
            response.EnsureSuccessStatusCode();
            string responseBody = await response.Content.ReadAsStringAsync();
            Resolver.Log.Info(responseBody);
        }
        catch (TaskCanceledException)
        {
            Resolver.Log.Info("Request time out.");
        }
        catch (Exception e)
        {
            Resolver.Log.Info($"Request went sideways: {e.Message}");
        }
    }
}
```

duduita, 2024-03-26:
Please ensure that you are connected before making the HTTP request using Ethernet, since the OS may take a few seconds to establish a connection.

```
public override async Task Run()
{
    Resolver.Log.Info("Run...");

    var ethernet = Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();

    ethernet.NetworkConnected += EthernetAdapterNetworkConnected;

    if (ethernet.IsConnected)
    {
            await GetWebPageViaHttpClient("https://postman-echo.com/get?foo1=bar1&foo2=bar2");
    }
}
```

As can be seen in this [Ethernet sample app](https://github.com/WildernessLabs/Meadow.Samples/blob/main/Source/Meadow.Core.Samples/Source/Network/Ethernet_Basics/MeadowApp.cs).

ctacke, 2024-03-26:
Tried the same app with 1.10 and http took 3.5 minutes.  The sample actually is checking that it is connected, I just didn't put it in the code above.

duduita, 2024-03-26:
Interesting, here the first request takes ~30s, and then the others take ~3s
```
Meadow StdOut: Initializing OS...
Meadow StdOut: Parsing app.config.yaml...
Meadow StdOut: Log level: Trace
Meadow StdOut: Looking for app assembly...
Meadow StdOut: Found '/meadow0/App.dll'
Meadow StdOut: Looking for IApp...
Meadow StdOut: Device is configured to use Wired Ethernet for the network interface
Meadow StdOut: Interface: eth0: eth0 Ethernet Up
Meadow StdOut: Creating 'F7CoreComputeV2' instance took 2725ms
Meadow StdOut: Device Initialize starting...
Meadow StdOut: Device is configured to use Ethernet for the network interface
Meadow StdOut: Interface: eth0: eth0 Ethernet Up
Meadow StdOut: PlatformOS Initialize starting...
Meadow StdOut: Ethernet InvokeEvent NetworkConnectedEvent returned CompletedOk
Meadow StdOut: Ethernet connected event triggered!
Meadow StdOut: SD Card is configured to 'off'
Meadow StdOut: File system Initialize starting...
Meadow StdOut: Creating instance of MeadowApp...
Meadow StdOut: Creating 'MeadowApp' instance took 74ms
Meadow StdOut: All cloud features are disabled.
Meadow StdOut: Initializing App
Meadow StdOut: Running App
Meadow StdOut: Run...
Meadow StdOut: lo
Meadow StdOut: ==
Meadow StdOut: Adapter name: lo
Meadow StdOut: Interface type .......................... : Loopback
Meadow StdOut: Physical Address ........................ :
Meadow StdOut: Operational status ...................... : Up
Meadow StdOut: IP version .............................. : IPv4
Meadow StdOut: MTU ..................................... : 0
Meadow StdOut: eth0
Meadow StdOut: ====
Meadow StdOut: Adapter name: eth0
Meadow StdOut: Interface type .......................... : Ethernet
Meadow StdOut: Physical Address ........................ : 22774D10127A
Meadow StdOut: Operational status ...................... : Up
Meadow StdOut: IP version .............................. : IPv4
Meadow StdOut: MTU ..................................... : 0
Meadow StdOut: IP address .............................. : 192.168.18.43
Meadow StdOut: Subnet mask ............................. : 255.255.255.0
Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 03/26/2024 18:47:24
Meadow StdOut: Updater State -> Disconnected
Meadow StdOut: {
Meadow StdOut: "args": {
Meadow StdOut: "foo1": "bar1",
Meadow StdOut: "foo2": "bar2"
Meadow StdOut: },
Meadow StdOut: "headers": {
Meadow StdOut: "x-forwarded-proto": "https",
Meadow StdOut: "x-forwarded-port": "443",
Meadow StdOut: "host": "postman-echo.com",
Meadow StdOut: "x-amzn-trace-id": "Root=1-66031852-4f6ab76b534321f4641455dd"
Meadow StdOut: },
Meadow StdOut: "url": "https://postman-echo.com/get?foo1=bar1&foo2=bar2"
Meadow StdOut: }
Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 03/26/2024 18:47:53
Meadow StdOut: {
Meadow StdOut: "args": {
Meadow StdOut: "foo1": "bar1",
Meadow StdOut: "foo2": "bar2"
Meadow StdOut: },
Meadow StdOut: "headers": {
Meadow StdOut: "x-forwarded-proto": "https",
Meadow StdOut: "x-forwarded-port": "443",
Meadow StdOut: "host": "postman-echo.com",
Meadow StdOut: "x-amzn-trace-id": "Root=1-6603185b-31e50edf593508d133818559"
Meadow StdOut: },
Meadow StdOut: "url": "https://postman-echo.com/get?foo1=bar1&foo2=bar2"
Meadow StdOut: }
Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 03/26/2024 18:47:55
Meadow StdOut: {
Meadow StdOut: "args": {
Meadow StdOut: "foo1": "bar1",
Meadow StdOut: "foo2": "bar2"
Meadow StdOut: },
Meadow StdOut: "headers": {
Meadow StdOut: "x-forwarded-proto": "https",
Meadow StdOut: "x-forwarded-port": "443",
Meadow StdOut: "host": "postman-echo.com",
Meadow StdOut: "x-amzn-trace-id": "Root=1-6603185d-022985ea56869f9443b134f5"
Meadow StdOut: },
Meadow StdOut: "url": "https://postman-echo.com/get?foo1=bar1&foo2=bar2"
Meadow StdOut: }
Meadow StdOut: Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 03/26/2024 18:47:57
```

ctacke, 2025-03-18:
Likely fixed, but should have a re-test to verify.