WildernessLabs/Meadow_Issues #553: Mlx90640 Driver and sample do not work as expected

DarkNoir, 2024-03-26:
**Describe the bug**
I am trying to use a Mlx90640 camera and its driver. Things are not making sense. Firstly, the array of floats returned by ReadRawData() represent what? The code for the sample makes it seem like degrees Fahrenheit because thresholds for ascii translation are between 68-98.6. My camera is returning values 30-24 so are not inside that range at all. (If my values are Celsius then its closer) but my values go up when things are colder and down when warmer (Ice pack reads 41.12 ish.  Querying other settings like GetMode() work as expected.  

Using this camera:
[https://www.amazon.com/MLX90640-Interface-Compatible-Raspberry-MLX90640-D55/dp/B07ZM[…]=app_do%3Aamzn1.fos.18ed3cb5-28d5-4975-8bc7-93deae8f9840&th=1](https://www.amazon.com/MLX90640-Interface-Compatible-Raspberry-MLX90640-D55/dp/B07ZMP995T/ref=sr_1_1?crid=1J9MHDH3OYUTQ&keywords=MLX90640-D55&qid=1697824563&sprefix=mlx90640-d55%2Caps%2C141&sr=8-1&ufe=app_do%3Aamzn1.fos.18ed3cb5-28d5-4975-8bc7-93deae8f9840&th=1)

**To Reproduce**
Steps to reproduce the behavior:
Run the sample and not the values returned are not aligned with the values expected in code. 

**Expected behavior**
Expected the sample and driver to align with the values returned by camera

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Meadow.Foundation (please complete the following information as best as you can):**
 - NuGet package version: 1.3.4

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 
 - IDE and version: Microsoft Visual Studio Enterprise 2022 (64-bit)  Version 17.7.6

 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.3.4

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2B-00-46-00-07-51-38-34-35-31-38-33
    Serial number: 337E316A3438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F5:FC

Firmware Versions
    OS: 1.3.4.0
    Mono: 1.3.4.0
    Coprocessor: 1.3.4.0
    Protocol: 7

**Additional context**

ctacke, 2024-10-15:
@Adrian, did this get fixed with the SPIComms ReadRegister fix?

adrianstevens, 2026-02-27:
This is fixed - closing issue

closed 2026-02-27 by adrianstevens