WildernessLabs/Meadow_Issues #563: Meadow not reconnecting to wifi

doingnz, 2024-03-27:
**Describe the bug**
Meadow does not automatically reconnect to Wifi.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the sample app [here](https://github.com/doingnz/Clima/tree/start-clima-azure-demo) 
2. Configure the correct wifi configuration in yaml
3. Deploy/debug the app and wait for app to connect to wifi and display IP address and the time
4. Switch off wifi at the router
5. Confirm app gets notification `Access point not found.`
6. Switch on the wifi and confirm there is no notification app reconnects to wifi

**Expected behavior**
Application gets notification of reconnect. NTP notifies of time update. 

**Screenshots**
example application output
```
IP Address is currently 192.168.2.117 & IsConnected=True
Time (UTC): 2024-03-27T18:30:41+00:00
Run GetTotalMemory= 905728
IP Address is currently 192.168.2.117 & IsConnected=True
Time (UTC): 2024-03-27T18:30:51+00:00
Run GetTotalMemory= 905760
IP Address is currently 192.168.2.117 & IsConnected=True
Wifi InvokeEvent NetworkDisconnectedEvent returned CompletedOk
--------------------------------------------------
Wifi_NetworkDisconnected:  Reason=Access point not found.
--------------------------------------------------
WiFi adapter state changed to: Disconnected
Time (UTC): 2024-03-27T18:31:01+00:00
Run GetTotalMemory= 906672
IP Address is currently 192.168.2.117 & IsConnected=False
Time (UTC): 2024-03-27T18:31:11+00:00
Run GetTotalMemory= 906672
IP Address is currently 192.168.2.117 & IsConnected=False
Time (UTC): 2024-03-27T18:31:21+00:00
```

**Developer tools :**
 - OS and version: Microsoft Windows [Version 10.0.22631.3374]
 - IDE and version: Visual Studio Enterprise 2022 v17.9.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.9.4

**Meadow:**
```
Getting device info...
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: MeadowOTA

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2E-00-36-00-03-51-38-34-35-31-38-33
    Serial number: 336E316D3438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D5:26:E8

Firmware Versions
    OS: 1.10.0.2
    Runtime: 1.10.0.2
    Coprocessor: 1.10.0.1
    Protocol: 8
```

**Additional context**
I am testing with Linksys router Archer AX11000 firmware 1.3.4 Build 20230906 rel 74809. 
The initial connect to wifi on boot is usually quite quick, so I think Meadow is Ok with this router and firmware.
```
Initialise Update Service.
Hello, Meadow Core-Compute!
Run Task.CompletedTask()
Time (UTC): 2024-03-27T09:50:06+00:00
Run GetTotalMemory= 648656
IP Address is currently 255.255.255.255 & IsConnected=False
Time (UTC): 2024-03-27T09:50:16+00:00
Run GetTotalMemory= 649048
IP Address is currently 255.255.255.255 & IsConnected=False
Time (UTC): 2024-03-27T09:50:26+00:00
Run GetTotalMemory= 649712
IP Address is currently 255.255.255.255 & IsConnected=False
Wifi InvokeEvent NtpUpdateEvent returned CompletedOk
--------------------------------------------------
TimeChanged (UTC): 2024-03-27T09:50:35+00:00
--------------------------------------------------
Wifi InvokeEvent NetworkConnectedEvent returned CompletedOk
WiFi adapter state changed to: Connected
Interface: wlan0: wlan0 Wireless80211 Up
--------------------------------------------------
IP Address is 192.168.2.117
--------------------------------------------------
Time (UTC): 2024-03-27T09:50:37+00:00
```

doingnz, 2024-03-27:
not clear if this is helpful, but the contents of `meadow.log` after the above test were as follows.
```C:\Temp>type meadow.log
I~2024-03-27T18:59:13Z~Logging started
I~2024-03-27T18:59:13Z~Config file successfully processed
I~2024-03-27T18:59:14Z~WiFi credentials file not found
```
Obviously the wifi credentials are available as Meadow did initially connect to the wifi.

NevynUK, 2024-03-28:
The way the OS is designed is as follows:

* There is a `MaximumRetryCount` in the `CoProcessor` section of the `meadow.config.yaml` file
* On disconnect, the ESP32 will try to reconnect to the access point and it will increment the retry count if, and only if, the retry count does not exceed the maximum
* If the retry count reaches the maximum then a `Disconnect` event should fire and automatic reconnect will cease
* If the `MaximumRetryCount` is not set then a default value of 3 will be used

From what I can see, the retry count is not set so the default of 3 will be used.

Also, I cannot see a subscription to a disconnect event in the sample application.

NevynUK, 2024-03-28:
Regarding the `WiFi credentials file not found` message.

The way the system is designed is that the WiFi config file will be read at startup and the entries transferred over to secure storage.  The file will then be deleted.  The board will now start the application.

The second time the board restarts, the file will not be on the board and the stored credentials will be used.

So this message is expected on the second and subsequent restarts until an application is redeployed to the board.

doingnz, 2024-03-29:
Apologies, I pasted incomplete URL. The sample app is off the `Additional Samples` folder and called [Clima_Azure_Demo ](https://github.com/doingnz/Clima/tree/start-clima-azure-demo/Source/Additional%20Samples/Azure/Clima_Azure_Demo ). It has nothing in it related to Clima at this stage as I went back to the most basic app possible that connects to the network in an effort to have a starting point with reliable & automatically recoverable network functionality given it needs to work at the bottom end of the garden and i am not going to walk down there to reboot the device anytime the wifi is disrupted. 

`public override Task Initialize()` includes 
```
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.NetworkConnected += Wifi_NetworkConnected;
            wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
            wifi.NetworkError += Wifi_NetworkError;
```

doingnz, 2024-03-29:
Please confirm I understand correctly OS Design is such that:
* Meadow OS will NOT attempt to reconnect to Wifi after it has fired the disconnect event.
* Meadow App must implement it's own logic to cope with reconnecting to wifi.

For reference the test app has the following defined. 
```
# Control how the ESP coprocessor will start and operate.
Coprocessor:

    # Should the ESP32 automatically attempt to connect to an access point at startup?
    # If set to true, wifi.yaml credentials must be stored in the device.
    AutomaticallyStartNetwork: true

    # Should the ESP32 automatically reconnect to the configured access point?
    AutomaticallyReconnect: true

    # Maximum number of retry attempts for connections etc. before an error code is returned.
    MaximumRetryCount: 7
```

doingnz, 2024-03-31:
Is it possible to power off the wifi functionality in the ESP32 in current Meadow firmware & hardware to save battery power?

If there is control capability and If the battery capacity and Wifi power draw is a concern, there needs to be a way to power on the wifi, connect, send / receive data by the app, disconnect from wifi, power off the wifi.  

Are all the steps of that workflow possible and implemented with the current design? e.g. maybe it is not possible to power off the wifi circuitry. 

The situation where the wifi network is not reachable is a variation on the above. Either it's not reachable when the app tries to connect, or it's a forced disconnect when already connected and the wifi becomes unreachable (dies, out of range, interference, ...).

This is why my sample app has a sample state-machine class than needs to be fleshed out to respond in a defined manner to the possible permutations in network connectivity.  There are similar issues with NTP as some functionality like logging may only make sense if the data is time stamped correctly.

I would not expect many people would opt for the 500,000 minutes + reset approach. That gives up all control and feedback. Would prevent switching to alternate backup network (cell?) another mechanisms to indicate wifi unavailable(flash RGB LED) . 

One place to start is the Meadow IoT accelerator projects and how end users (not C# devs) buying a finished product (e.g. Weather Station) might need it to work and if a developer buying the Meadow kit can make such a finished product. i.e. what Meadow OS support needed to achieve all the functionality for a finished product.

closed 2025-04-08 by halyssonJr