WildernessLabs/Meadow_Issues #564: Meadow.Foundation - Displays.WinForms not rendering in right dimentions

jorgedevs, 2024-03-28:
In v1.10.0, running a Meadow.Desktop project creates a Project Lab display size WinForms display (320x240). However, that dimention is set as the full window size, so part of the actual buffer is clipped outside of the Window:

[image]

On my machine (probably not for all monitors/PCs), I calculated that right and left window borders are 8 pixels, and bottom is 8 pixels and top with the window controls and title is 31 pixels, so dimensions should be:

Width: 320 + 8 + 8 =  336
Height: 240 + 8 + 31 = 279

[image]

Repro project attached.
[Project2.zip](https://github.com/WildernessLabs/Meadow_Issues/files/14795312/Project2.zip)

adrianstevens, 2024-03-28:
Fixed in develop

ctacke, 2024-05-01:
Fixed in 1.11

closed 2024-05-01 by ctacke