WildernessLabs/Meadow_Issues #566: Meadow.Cloud - Sending Health Metrics before logging in to Meadow.Cloud at startup

jorgedevs, 2024-04-02:
When I enable Health Metrics in 30 minute Intervals and I run the app on a FeatherF7:

```console
Initialize...
Wifi InvokeEvent NtpUpdateEvent returned CompletedOk
Wifi InvokeEvent NetworkConnectedEvent returned CompletedOk
WiFi adapter state changed to: Connected
Running App
Run...
Interface: wlan0: wlan0 Wireless80211 Up
starting health metrics timer 
sending health metric failed                                   <================= SENDING HEALTH METRIC FAILS
CloudConnectionState: Disconnected
Meadow.Cloud service will use the Esp32WiFiAdapter network interface
CloudConnectionState: Authenticating
Attempting to login to https://www.meadowcloud.co/api/devices/login with {"id":"2D-00-20-00-09-50-30-52-54-31-35-20"}...
authentication successful. extracting token
decrypting auth payload
auth token successfully received
Authentication took 34,299.00 ms
CloudConnectionState: Connecting
Creating MQTT client options
Adding MQTT creds
Connecting MQTT client
MQTT connected
CloudConnectionState: Subscribing
Meadow.Cloud service subscribing to 'd82f8ec7243a461581c2544070209fbd/commands/2D-00-20-00-09-50-30-52-54-31-35-20'
Meadow.Cloud service subscribing to 'd82f8ec7243a461581c2544070209fbd/ota/2D-00-20-00-09-50-30-52-54-31-35-20'
UpdateState Connected
Updater State -> Connected
CloudConnectionState: Connected
````

You can see in that output that it tried to send health metrics before login in Meadow.Cloud. Does that mean I need to wait 30 mins for the next logs to be sent?

I would prefer it it'll wait to log in to Meadow.Cloud before trying to send the logs (or check periodically), id like a baseline logs as soon as the app starts.

doingnz, 2024-04-02:
Please see discussion in issue #563 with a related problem of reconnecting to wifi should wifi connectivity be lost after initial connection.  There needs to be a strategy to manage connecting to a network, getting time, login to cloud, logging metrics, check for updates, accept notification of network disconnect, poll for network availability, reconnect to network and the above services.

lamebrain, 2024-04-23:
@jorgedevs can you try with latest? With the new store/forward, this should work as expected.

jorgedevs, 2024-04-23:
Trying now

jorgedevs, 2024-04-23:
> Please see discussion in issue #563 with a related problem of reconnecting to wifi should wifi connectivity be lost after initial connection. There needs to be a strategy to manage connecting to a [...]

Thanks @doingnz , there will be upcoming changes in network events that will deal with those scenarios.

doingnz, 2024-04-23:
linking to the related #591 & #593

jorgedevs, 2024-04-23:
This will work regardless of the other issues, and this is working as of last official release, so I'm closing this one for now.

closed 2024-04-23 by jorgedevs