WildernessLabs/Meadow_Issues #567: Cultivar crashes with error and doesn't come back up.

bryancostanich, 2024-04-02:
**Describe the bug**
Cultivar crashes with `Failed to send Meadow.Cloud data: The SSL connection could not be established, see inner exception.` and after some number of those failures, it goes down and doesn't come back up. I see this in Meadow.Cloud

**To Reproduce**
Steps to reproduce the behavior:
1. Clone `Cultivar` and run on a Project Lab (edit wifi creds)
2. watch on Meadow.Cloud
3. It should crash with that error at some point.

**Expected behavior**
Runs without error. If it does error out and crash, device should reset.

**Developer tools (please complete the following information as best as you can):**
n/a

**Meadow (please complete the following information as best as you can):**
```
Board Information 
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: Cultivar

Hardware Information 
    Processor type: STM32F777IIK6
    ID: 36-00-2E-00-07-51-38-34-35-31-38-33
    Serial number: 336631753438
    Coprocessor type: ESP32
    MAC Address - 
	WiFi: 4C:75:25:D5:28:78

Firmware Versions 
    OS: 1.10.0.2
    Runtime: 1.10.0.2
    Coprocessor: 1.10.0.1
    Protocol: 8
```

bryancostanich, 2024-04-23:
Handshake issue likely.

duduita, 2024-05-22:
Some additional info, using the latest release, and enabling some native logs, we can see a more descriptive error message:
```
Meadow StdOut: mbedtls_net_recv: read returned -1
Meadow StdOut: Exception sending cloud message
Meadow StdOut: The SSL connection could not be established, see inner exception.(-76)
Meadow StdOut: TLS handshake failed. Return code: -76. Error: Reading information from the socket failed.(-76)
```
As can be seen in the following logs, for some reason suddenly the underlying read method starts to return -1, but it keeps going, until getting a new token, after that, all the requests fail with the -76 error repetitively, then we get an OOM error, which I suspect that might be related to this other [issue](https://github.com/WildernessLabs/Meadow_Issues/issues/507), regarding a possible memory issue.

```
Meadow StdOut: making cloud request to /api/events with payload: {"measurements":{"health.cpu_temp_celsius":47.8571428571429,"health.memory_used":1366680,"health.disk_space_used":14029703,"info.os_version":"1.11.1.13","info.coprocessor_os_version":"1.11.0.0"},"description":"device.health","eventId":10,"timestamp":"2024-05-17T23:34:57.735Z"}
Meadow StdOut: cloud request to /api/events completed successfully
Meadow StdOut: Temperature: 39.0 | Humidity: 32.2 | Moisture: 2.3
Meadow StdOut: Bearer: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6IjZmZDJiMWNmODRhZTRhMjE4MDA4ZTMxNGFlMzVjNmVhIn0.eyJzdWIiOiIyRS0wMC0yNS0wMC0xMi01MC0zMy01My0zMi0zNC0zMS0yMCIsIm9pZCI6IjM2NTE5Y2M5YTYwYzRkMjg4ZTQ5ZGZlZjAwNDY0ZTc2IiwibmJmIjoxNzE1OTg1MzAyLCJleHAiOjE3MTU5ODg5MDIsImlzcyI6Imh0dHBzOi8vbWVhZG93Y2xvdWQuY28iLCJhdWQiOiJtZWFkb3ctY2xvdWQtZGV2aWNlIiwibXF0dC1hY2wiOnsic3ViIjpbIjM2NTE5Y2M5YTYwYzR
Meadow StdOut: Mjg4ZTQ5ZGZlZjAwNDY0ZTc2LyMiXSwicHViIjpbIjM2NTE5Y2M5YTYwYzRkMjg4ZTQ5ZGZlZjAwNDY0ZTc2LyMiXX19.GaEO2xP-QdDCEbVUGKhUVb7fAi6vKbUogeRtGqW99aQZ9_QKlWAEXiOpyNOlrTi2jXqXxFZTjf2KQ0hyTz4hJwKOEImbR8YBGAooxSxtMEKDWwOvYlLkX8Pa58vpZSyqr_BKvOQ8Z_5Xk1hIQ8vtdPWLzVeq6OXahoeiSbDm7yWJbykdyfK5AtFD5jn1trRm8rtbAWL1Ne2GZIueUnEBpO-2oGrZWCxSO8ShXlVzNv1tkSRZzFvQSIwsF-l4X4HiesIgK7GC9TIiLOO0VxXVZee0nVdM5zz0ok
Meadow StdOut: p6yvbTKtTkblxkmSxOSk16S08Ex223DiXONRm_5uVqXoqeGiCXQ
Meadow StdOut: making cloud request to /api/events with payload: {"measurements":{"LogId":1277,"TemperatureCelsius":39.015625,"HumidityPercent":32.226852805486,"SoilMoistureDouble":2.32579550158359},"description":"Atmospheric reading","eventId":110,"timestamp":"2024-05-17T23:35:15.633Z"}
Meadow StdOut: Attempting to login to https://staging.meadowcloud.dev/api/devices/login with {"id":"2E-00-25-00-12-50-33-53-32-34-31-20"}...
Meadow StdOut: authentication successful. extracting token
Meadow StdOut: decrypting auth payload
Meadow StdOut: auth token successfully received
Meadow StdOut: Bearer: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6IjZmZDJiMWNmODRhZTRhMjE4MDA4ZTMxNGFlMzVjNmVhIn0.eyJzdWIiOiIyRS0wMC0yNS0wMC0xMi01MC0zMy01My0zMi0zNC0zMS0yMCIsIm9pZCI6IjM2NTE5Y2M5YTYwYzRkMjg4ZTQ5ZGZlZjAwNDY0ZTc2IiwibmJmIjoxNzE1OTg4OTIwLCJleHAiOjE3MTU5OTI1MjAsImlzcyI6Imh0dHBzOi8vbWVhZG93Y2xvdWQuY28iLCJhdWQiOiJtZWFkb3ctY2xvdWQtZGV2aWNlIiwibXF0dC1hY2wiOnsic3ViIjpbIjM2NTE5Y2M5YTYwYzR
Meadow StdOut: Mjg4ZTQ5ZGZlZjAwNDY0ZTc2LyMiXSwicHViIjpbIjM2NTE5Y2M5YTYwYzRkMjg4ZTQ5ZGZlZjAwNDY0ZTc2LyMiXX19.XkxdowMraKGbfi1ni5yp8uJCrtH7mJPhBpswADvPxWe__7bsBCVhB6GrAyOBmKyxcbVVIC-SJc-x3u6CUIHmdSnCb8nd4TxbaJVKPEjz2Drg4xmguKfUg8G4uk8dTrhYvM6SJZJsJzH27vXKiB_zUPrgpu7XgbTIFajC9B6392cva1vsbGJ0ekQv8I3sh_CC4WW0WloYEOcou-DroC--oGlIFGorKIzD728qxBiya_cgJTDnGCsSWqsWRUmveY5lnvpizrXHp9O-Ca3f52q_VyHqSgk_pSriq0
Meadow StdOut: E4nHNB_FDpPI-esQsj_dad9AhGTyxp3CfYWPnAusbC20El2aSzg
Meadow StdOut: making cloud request to /api/events with payload: {"measurements":{"LogId":1277,"TemperatureCelsius":39.015625,"HumidityPercent":32.226852805486,"SoilMoistureDouble":2.32579550158359},"description":"Atmospheric reading","eventId":110,"timestamp":"2024-05-17T23:35:15.633Z"}
Meadow StdOut: mbedtls_net_recv: read returned -1
Meadow StdOut: Exception sending cloud message
Meadow StdOut: The SSL connection could not be established, see inner exception.(-76)
Meadow StdOut: TLS handshake failed. Return code: -76. Error: Reading information from the socket failed.(-76)
Meadow StdOut: CLOUD ERROR: Exception sending cloud message
```

closed 2024-06-18 by bryancostanich