WildernessLabs/Meadow_Issues #570: Power connection to MeadowF7 via INA219 breakout chip will not create WiFi Connection

kingdona, 2024-04-02:
**Describe the bug**
I'm having an issue with getting WiFi to connect when running my application on Meadow that is powered via the INA219 breakout chip. Everything else works fine including the power monitoring and display. I'm not sure if for some reason the .config files aren't being loaded when it boots? When I connect power directly to the meadow it works fine.

[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
Connect Meadow using the setup in the attached pictures and then configure Wifi via the YAML files. If the Meadow is powered by its onboard Micro-USB connector, the Wifi works. When it is powered by an external power source via the power rails on a breadboard, other peripherals will work but the Wifi will not connect.

**Expected behavior**
I expect the Wifi to work regardless of how the Meadow is being powered.

**Screenshots**
Pictures attached.

**Developer tools (please complete the following information as best as you can):**
Windows 11 Pro 23H2 OS Build 22631.3374
Microsoft Visual Studio Community 2022 (64-bit) - Current Version 17.9.5
Meadow extension for IDE version: 1.9.4

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI]Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7
Firmware Versions
    OS: 1.10.0.2
    Runtime: 1.10.0.2
    Coprocessor: 1.10.0.1
    Protocol: 8 

**Additional context**

kingdona, 2024-04-02:
Chris: I wonder if the power supply of a meadow isn't enough to fully power two meadows (plus your peripherals) and the ESP32 on the second one (which is where the WiFi transceiver is) is browning out.  Just a guess.

Don: I was planning to get an independent power supply to power this. I'll go ahead and do that and give it a try. Thanks!

Don: Tried with a separate dedicated 5V, It reads 5.19v on my multimeter and I get same reading on meadow 5V and ground.
Still same result, never gets the WiFi connection.

Chris: Maybe 
Mark has ideas?

Mark: My first instinct is that the regulator is not able to supply enough current to run everything.  I have just looked at the schematic and the regulator is the LDL212PV33R which according to the datasheet for the LDL212 from ST should be good to 1.2A.  So I would say that the limit may be the USB supply to the system.  If this is A to Micro-C then we could be seeing the limit of 500mA for the system.  This is certainly too low for two meadows plus the additional circuitry you have connected.  Meadow peaks at about 250-300 mA on startup when WiFi is being used.

Mark: It settles down after startup and power requirement will depend upon application.  I have run a board making continuous WiFi requests and the average is about 180mA power draw.  To be clear that is constant WiFi requests so we expect that scenario to be power hungry.

Mark: Have you have tested that both boards run OK when not sharing a power supply ?

Don: There is actually only one Meadow board in this setup. I originally was powering the setup with another Meadow board and then put together the independent USB power supply.

Don: The power connection is a USB-A coming from my monitor. Not sure how to measure mA but if that might be the issue, I could switch it to USB-C to C.

Mark: Just to be clear, you have one Meadow with the INA219 and a display on bread board.

Mark: The second board has external power connected to the power rails of the breadboard through to the 5V line of the Meadow and that is not working.

Mark: In both cases, is the power supply from the same source - the monitor USB ?

Don: In the first image where the power supply is directly into the Micro-USB connection of the Meadow, that is coming from my Laptop via a USB-C connection and it works. The second image with the power on the breadboard rails is being powered from a USB-A coming from my monitor and it does not work.
To be clear, I am referring to only the WiFi when talking about whether a setup works or doesn't work. That is my main issue.
I suppose the other obvious thing to mention is that the power connection via the breadboard is passing through the INA219 breakout. I assume it is set up correctly and I did have it working properly providing power information on the display. I am not sure how much of a factor it is here with the issue I am having

Don: I have an AdaFruit 1833 USB Micro-B breakout board coming to see if it makes any difference being powered via USB-C from my laptop. I hope to have those results later today.

kingdona, 2024-04-02:
Same results using an Adafruit 1833 Micro-USB breakout board powered via USB-C.

engunneer, 2024-04-03:
I looked at the Adafruit schematics, and the INA219 has 10K pull-up resistors on SDA and SCL, while the OLED breakout has 2.2k pull-up resistors, so depending on what the Meadow has for pull-up configuration, SDA and SCL are probably around 3.7 volts, which probably won't damage things but could potentially interfere with the Meadow? The other option is that the wifi has intermittent need for a burst of power and there may not be enough capacitance on the wifi's power rail to keep it happy, since the INA219 might be acting like a filter (even though the resistor on the INA219 is quite small)

kingdona, 2024-04-03:
The original goal here was to be able to monitor power consumption for a project I am working on where I need a 3.7v 2200mAh battery to be able to last for 3-4 weeks without a recharge as the device will be encased and submerged in liquid. I know what my power budget is per hour and so I needed to know what it would cost me to wake the OS, take a reading from a combo temperature/accelerometer featherwing, connect to Wifi, send data to a service and then go back to sleep.

When I added connecting to Wifi, that is when I ran into this issue. The display was just to be able to display the readings from the INA219 chip for various operations. I am a novice at this so I don't really know if or how the screen affects the readings. I'm just getting started with the process. I'll give your suggestion a try and let you know how it goes. Thanks!

kingdona, 2024-04-03:
Same result. Attached a picture to make sure I am not misunderstanding your suggestion.
[image]

engunneer, 2024-04-11:
sorry about the delay on this. I misplaced my INA219 and got another this week. I was able to reproduce your setup, but not the issue. I think the only difference is that I have a v2.d F7 instead of your v2.c.

[image]

Running with the USB power to the Meadow or to the external connection, I get wifi connection information. For whatever reason, the USB directly to the meadow usually takes a little longer to make the connection (18-40 seconds after the init starts), versus the other way (7-10 seconds). This is not rigorously tested, but seemed interesting.
I've published [my test code](https://github.com/engunneer/Meadow-InaWifiTest) as well. I'm using 1.10.0.2 firmware, and nuget published packages v1.10 for the hardware bits. (Incidentally, I'm the author of the INA2xx driver in it's current form).

kingdona, 2024-04-12:
Ok, got it to work. Thanks much for your help. The code was very helpful as well and I might have had something off there that was a contributing factor. I am going to go back and review. I definitely had some wiring incorrect and tbh, I am still not entirely sure I understand completely how it works and so I am going to go back over it, review it and try to recreate it again so I understand. Thanks again for your help.

[image]