WildernessLabs/Meadow_Issues #572: Meadow templates aren't auto-installing on VS2022 on Windows

adrianstevens, 2024-04-03:
Fresh install, installed the extension, worked on a couple of non-Meadow projects, restarted Visual Studio several times.

Tried to create a Meadow project 2 days later and no templates were installed.

I had to install them manually and restart Visual Studio to create a new project.

CartBlanche, 2024-04-03:
@adrianstevens This is what I did:
1. I uninstalled the `1.9.6` extension from Community, Professional, Enterprise (all these are on `17.9.5`) and Preview (`17.10.0 preview 2`)
2. I uninstalled the templates doing `dotnet new uninstall WildernessLabs.Meadow.Template`
3. I then installed the `1.9.6` extension just into Community and launched it.
4. After the IDE has started I then ran the following command  `dotnet new uninstall` which then lists all templates that can be uninstalled and I got.
 ```
dotnet new uninstall
Currently installed items:
   WildernessLabs.Meadow.Template
      Version: 1.10.0
      Details:
         Author: Wilderness Labs
         Owners: Meadow
         Reserved: ✘
         NuGetSource: https://api.nuget.org/v3/index.json
      Templates:
         Meadow Core-Compute App (CoreComputeModule) C#
         Meadow Core-Compute App (CoreComputeModule) F#
         Meadow Core-Compute App (CoreComputeModule) VB.NET
         Meadow.Desktop App (MeadowDesktop) C#
         Meadow F7 Feather App (F7Feather) C#
         Meadow F7 Feather App (F7Feather) F#
         Meadow F7 Feather App (F7Feather) VB.NET
         Meadow.Linux Jetson Nano App (JetsonNano) C#
         Meadow Library (Library) C#
         Meadow Library (Library) F#
         Meadow Library (Library) VB.NET
         Meadow Project Lab App (ProjectLab) C#
         Meadow.Linux Raspberry Pi App (RaspberryPi) C#
         Meadow.Linux reTerminal App (reTerminal) C#
      Uninstall Command:
         dotnet new uninstall WildernessLabs.Meadow.Template
```
5. In VS2022 Community I did File New and filtered on Meadow and the templates appeared there.
6. While the IDE was running, in a command window I ran `dotnet new uninstall WildernessLabs.Meadow.Template`
7. In VS2022  I did File New and filtered on Meadow and zero Meadow templates appeared there.
8. I shutdown and restarted VS2022,  then did File New and filtered on Meadow and the templates appeared there again.

Which version of the extension did you have installed?

ctacke, 2025-01-07:
Template installation is a separate thing from installing the extensions.  Let's not try doing too much with the limited resources we have.

closed 2025-01-07 by ctacke