WildernessLabs/Meadow_Issues #576: Cloud logs/events need 2 times - server and device

ctacke, 2024-04-04:
My device send a crash log.  The log was sent before the device got an NTP time, so the record has a stale/default record time and ends up at the bottom of my log list.  There should be two (sortable!) columns - the time stored in the record *and* the time the server received it.

j-koda80, 2024-04-05:
@ctacke we actually do add a ReceivedAt Property to the Event/Logs Payload sent from the device before storing in Elasticsearch. But today we look at the Timestamp Property that the device provides as part of the payload for data purging. I can easily switch to the RecievedAt Property for the purging.

j-koda80, 2024-04-05:
@ctacke there are already 2 sortable columns in the UI Grid for 2 dates. Last 2 columns, second to last called ReceivedAt and last column called Timestamp, do you not see these columns for your device logs/events?

ctacke, 2024-04-06:
I see "Timestamp"
[image]

j-koda80, 2024-04-07:
I will update the UI to show ReceivedAt Column, it's currently ignoring the property.

adunndevster2, 2024-04-24:
@j-koda80 - This is all fixed and in production now?

closed 2024-05-14 by alexischr