WildernessLabs/Meadow_Issues #577: sending mailmessage via smtpclient. normalization errors

JT-50, 2024-04-04:
when creating a basic email in vb.net

the following code:
```
Dim mail As New MailMessage()
mail.From = New MailAddress("meadow@example.com")
```
            
fails with the error:
`The type initializer for 'System.Text.Normalization' threw an exception.`

I also saw this message when trying to change up my code:
`System.NotSupportedException: This runtime has been compiled without string normalization support.`

adrianstevens, 2024-04-16:
Thanks @JT-50 - we'll need to investigate, this may be a linker error

halyssonJr, 2025-01-20:
@JT-50 , could you share the piece of code related to this resource (**EMail Address**)? I have been working this issue and would like to re-test using an internal binary before closing this issue.

JT-50, 2025-01-20:
`Public Shared Sub CreateTestMessage()
    Try

        Dim [to] As MailAddress = New MailAddress("jt@micromsp.com")
        Dim from As MailAddress = New MailAddress("jt@micromsp.com")
        Dim message As MailMessage = New MailMessage(from, [to])
        message.Subject = "Using the new SMTP client."
        message.Body = "Using this new feature, you can send an email message from an application very easily."
        Dim client As SmtpClient = New SmtpClient()
        Console.WriteLine("Sending an email message to {0} at {1} by using the SMTP host={2}.", [to].User, [to].Host, client.Host)
        client.Send(message)

    Catch ex As Exception
        Resolver.Log.Info($"email not sent:{ex.ToString}")
    End Try
End Sub`

`[application] email not sent:System.TypeInitializationException: The type initializer for 'System.Text.Normalization' threw an exception. ---> System.NotSupportedException: This runtime has been compiled without string normalization support.
at (wrapper managed-to-native) System.Text.Normalization.load_normalization_resource(intptr&,intptr&,intptr&,intptr&,intptr&,intptr&)
at
ystem.Text.Normalization..cctor () [0x0001d] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
--- End of inner exception stack trace ---
at System.String.Normalize (System.Text.NormalizationForm normalizationForm) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
at System.Net.Mail.MailAddressParser.NormalizeOrThrow (System.String input) [0x00000] in <20f6f3020bb24d228c9455a67178a
34>:0
at System.Net.Mail.MailAddressParser.ParseDomain (System.String data, System.Int32& index) [0x00049] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddressParser.ParseAddress (System.String data, System.Boolean expectMultipleAddresses, System.Int32& index) [0x00029] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddressParser.ParseAdd
ess (System.String data) [0x00009] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddress..ctor (System.String address, System.String displayName, System.Text.Encoding displayNameEncoding) [0x000d7] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddress..ctor (System.String address) [0x00000] in <20f6f3020bb24d228c9455a67178a034>:0
at autoc
op.autocoop.MeadowApp.CreateTestMessage () [0x00002] in <d0f4e4d947cf41388128a5e29d082e4e>:0`

JT-50, 2025-01-20:
`
Public Shared Sub CreateTestMessage()
    Try

        Dim [to] As MailAddress = New MailAddress("jt@domain")
        Dim from As MailAddress = New MailAddress("jt@domain")
        Dim message As MailMessage = New MailMessage(from, [to])
        message.Subject = "Using the new SMTP client."
        message.Body = "Using this new feature, you can send an email message from an application very easily."
        Dim client As SmtpClient = New SmtpClient("smtphost", 25)
        client.Send(message)

    Catch ex As Exception
        Resolver.Log.Info($"email not sent:{ex.ToString}")
    End Try
End Sub
`

`[application] email not sent:System.TypeInitializationException: The type initializer for 'System.Text.Normalization' threw an exception. ---> System.NotSupportedException: This runtime has been compiled without string normalization support.
at (wrapper managed-to-native) System.Text.Normalization.load_normalization_resource(intptr&,intptr&,intptr&,intptr&,intptr&,intptr&)
at
ystem.Text.Normalization..cctor () [0x0001d] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
--- End of inner exception stack trace ---
at System.String.Normalize (System.Text.NormalizationForm normalizationForm) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
at System.Net.Mail.MailAddressParser.NormalizeOrThrow (System.String input) [0x00000] in <20f6f3020bb24d228c9455a67178a
34>:0
at System.Net.Mail.MailAddressParser.ParseDomain (System.String data, System.Int32& index) [0x00049] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddressParser.ParseAddress (System.String data, System.Boolean expectMultipleAddresses, System.Int32& index) [0x00029] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddressParser.ParseAdd
ess (System.String data) [0x00009] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddress..ctor (System.String address, System.String displayName, System.Text.Encoding displayNameEncoding) [0x000d7] in <20f6f3020bb24d228c9455a67178a034>:0
at System.Net.Mail.MailAddress..ctor (System.String address) [0x00000] in <20f6f3020bb24d228c9455a67178a034>:0
at autoc
op.autocoop.MeadowApp.CreateTestMessage () [0x00002] in <d0f4e4d947cf41388128a5e29d082e4e>:0`

halyssonJr, 2025-01-21:
@JT-50, good news, I tested a similar sample that you shared, and works. I will change the label from re-test to fix in the upcoming release. 
- C# Code ([here](https://gist.github.com/halyssonJr/d8abc65925a1fd1ee4cb265247c7893e))

[image]

JT-50, 2025-03-21:
Sorry i didnt get back sooner, i had removed emailling from my app since it hasnt been implemented.

This is a basic SMTP sent to a local SMTP server on my LAN and requires no authentication.

`13:53:09:332 [application] Email test failed: Error Initializing the configuration system. 13:53:10:029 [application] Stack Trace: at System.Configuration.ClientConfigurationSystem.get_Configuration () [0x0002a] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:029 at System.Configuration.ClientConfigurationSystem.System.Configuration.Internal.IInternalConfigSystem.GetSection (System.String configKey) [0x00000] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:029 at System.Configuration.ConfigurationM 13:53:10:030 nager.GetSection (System.String sectionName) [0x00005] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:030 at System.Net.Mail.SmtpClient..ctor (System.String host, System.Int32 port) [0x0001c] in <a57b9164a6134732b2e2c66a145fe1c7>:0 13:53:10:030 at autocoop.autocoop.MeadowApp.TestEmailDirectly () [0x0002b] in <cb95ff77e20549e3b5da17e295dcb04a>:0 13:53:10:790 [application] Email not sent: Error Initializing the configuration system. 13:53:10:790 [application] Exception Type: ConfigurationErrorsException 13:53:10:790 [application] Inner Exception: Default constructor not found for type System.Configuration.ExeConfigurationHost 13:53:10:790 [application] Stack Trace: at System.Configuration.ClientConfigurationSystem.get_Configuration () [0x0002a] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:790 at System.Configuration.ClientConfigurationSystem.System.Configuration.Internal.IInternalConfigSystem.GetSection (System.String configKey) [0x00000] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:790 at System.Configuration.ConfigurationM 13:53:10:790 nager.GetSection (System.String sectionName) [0x00005] in <97de0d4a33d940b0b57493e06b157f42>:0 13:53:10:790 at System.Net.Mail.SmtpClient..ctor (System.String host, System.Int32 port) [0x0001c] in <a57b9164a6134732b2e2c66a145fe1c7>:0 13:53:10:790 at System.Net.Mail.SmtpClient..ctor () [0x00000] in <a57b9164a6134732b2e2c66a145fe1c7>:0 13:53:10:790 at autocoop.autocoop.MeadowApp.CreateTestMessage () [0x0004b] in 13:53:10:790 <cb95ff77e20549e3b5da17e295dcb04a>:0`

**Copilot AI says:**

The "Error Initializing the configuration system" combined with the "Default constructor not found for type System.Configuration.ExeConfigurationHost" indicates that the Meadow runtime environment does not fully support System.Configuration, leading to issues when using SmtpClient. This is a critical limitation with the current implementation and the Meadow device's support for legacy .NET features.

Here’s how to adapt and resolve this issue:

Cause of the Issue
System.Configuration Dependency:

The SmtpClient class is attempting to access configuration settings, even when explicitly configured in code. This behavior ties back to its legacy design and reliance on System.Configuration.

Meadow's Limited Support:

Meadow devices do not support the full .NET Framework, so the required ExeConfigurationHost and related services are unavailable, causing the initialization error.

Legacy SmtpClient:

The SmtpClient class is considered outdated in modern .NET applications. Alternatives such as MailKit or System.Net.Http.HttpClient are more robust, lightweight, and configuration-independent.

**Code that causes this:**

  Public Shared Sub CreateTestMessage()
      Try
          ' Define sender and recipient email addresses
          Dim fromAddress As New MailAddress("chickencoop@domain.com", "Chicken Coop")
          Dim toAddress As New MailAddress("jt@domain.com", "JT")

          ' Create the email message
          Dim message As New MailMessage(fromAddress, toAddress) With {
      .Subject = "Daily Schedule Test Email",
      .Body = "This is a test email to confirm the SMTP setup is working as expected.",
      .IsBodyHtml = False ' Set this to True if you want to send an HTML email
  }

          ' Explicitly configure the SMTP client
          Using smtpClient As New SmtpClient() ' Avoid constructor with host and port so it doesn't invoke web.config or other non existent files
              smtpClient.Host = "192.168.0.20" ' Explicitly set the SMTP server address
              smtpClient.Port = 25           ' Set the SMTP port
              smtpClient.EnableSsl = False     ' Ensure SSL is disabled for your local server
              smtpClient.UseDefaultCredentials = True ' Do not use configuration file credentials

              ' Send the email
              smtpClient.Send(message)
          End Using

          ' Log success
          Resolver.Log.Info("Test email sent successfully.")
      Catch ex As Exception
          ' Log detailed error information for debugging
          Resolver.Log.Error($"Email not sent: {ex.Message}")
          Resolver.Log.Error($"Exception Type: {ex.GetType().Name}")
          If ex.InnerException IsNot Nothing Then
              Resolver.Log.Error($"Inner Exception: {ex.InnerException.Message}")
          End If
          Resolver.Log.Error($"Stack Trace: {ex.StackTrace}")
      End Try
  End Sub

halyssonJr, 2025-03-25:
Thanks @JT-50 for the updates. I will start this investigation. Did you have a chance to test this sample?
- https://github.com/WildernessLabs/Meadow.Samples/tree/develop/Source/Meadow%20F7/Network/Smtp

JT-50, 2025-03-25:
i am using vb.net

but the issue that AI found was that the smtp client was bound to some system.config file that doesn't exist in our meadow environment.

i was unable to send a basic SMTP email to a local MS SMTP server in my Lan.
And no, it's not the server,

I am now using TCPCLIENT - and can actually send email from my app using blocking ports and a "telnet like behavior" 
***IT MUST USE ASCII for some reason UTF-8 fails

` Public Shared Sub SendEmail()
     Try
         Dim smtpServer As String = "smtp.*******.local"
         Dim smtpPort As Integer = 2525

         Resolver.Log.Info("Connecting to SMTP server...")
         Using client As New TcpClient(smtpServer, smtpPort)
             client.Client.Blocking = True ' Enable blocking mode
             Using stream As NetworkStream = client.GetStream()
                 Dim writer As New StreamWriter(stream, Encoding.ASCII) With {.AutoFlush = True}
                 Dim reader As New StreamReader(stream, Encoding.ASCII)

                 ' Read the initial server banner
                 Dim banner As String = reader.ReadLine()
                 If String.IsNullOrEmpty(banner) Then
                     Resolver.Log.Error("No banner received from server.")
                     Exit Sub
                 End If
                 Resolver.Log.Info($"Server Banner: {banner}")

                 ' Prepare email content
                 Dim emailContent As New StringBuilder()
                 emailContent.AppendLine("<!DOCTYPE html>")
                 emailContent.AppendLine("<html lang='en'>")
                 emailContent.AppendLine("<head>")
                 emailContent.AppendLine("<meta charset='UTF-8'>")
                 emailContent.AppendLine("<style>")
                 emailContent.AppendLine("body { font-family: Verdana, sans-serif; background-color: #fef5d7; color: #333; margin: 0; padding: 0; }")
                 emailContent.AppendLine(".content { max-width: 600px; margin: 20px auto; padding: 20px; background: #fffbea; border-radius: 10px; border: 2px solid #f4d03f; }")
                 emailContent.AppendLine("h1 { text-align: center; color: #d35400; }")
                 emailContent.AppendLine("ul { padding: 0; list-style: none; }")
                 emailContent.AppendLine("ul li { padding: 8px 0; border-bottom: 1px dotted #ccc; }")
                 emailContent.AppendLine(".footer { text-align: center; font-size: 12px; color: #888; margin-top: 20px; }")
                 emailContent.AppendLine("</style>")
                 emailContent.AppendLine("</head>")
                 emailContent.AppendLine("<body>")
                 emailContent.AppendLine("<div class='content'>")
                 emailContent.AppendLine("<h1>Daily Chicken Coop Schedule</h1>")
                 emailContent.AppendLine("<p>Hello from the coop! Here's the schedule for today:</p>")
                 emailContent.AppendLine("<ul>")

                 emailContent.AppendLine("</ul>")
                 emailContent.AppendLine("<p>We'll be keeping things running smoothly. Have a great day!</p>")
                 emailContent.AppendLine("<div class='footer'>")
                 emailContent.AppendLine("Generated by *********************")
                 emailContent.AppendLine("</div>")
                 emailContent.AppendLine("</div>")
                 emailContent.AppendLine("</body>")
                 emailContent.AppendLine("</html>")

                 ' Build SMTP commands
                 Dim smtpCommands As String = $"EHLO ************l{vbCrLf}" &
                                      $"MAIL FROM:<********>{vbCrLf}" &
                                      $"RCPT TO:<jt@***********>{vbCrLf}" &
                                      $"DATA{vbCrLf}" &
                                      $"Subject: chicken coop{vbCrLf}" &
                                      $"MIME-Version: 1.0{vbCrLf}" &
                                      $"Content-Type: text/html; charset=UTF-8{vbCrLf}" &
                                      vbCrLf &
                                      emailContent.ToString() &
                                      vbCrLf &
                                      $".{vbCrLf}" &
                                      $"QUIT{vbCrLf}"

                 writer.Write(smtpCommands)
                 writer.Flush()
                 Resolver.Log.Debug("SMTP sent.")

                 ' Read and log the server's raw response
                 ' Dim buffer(1024) As Byte ' Initialize the buffer to read raw responses
                 ' Dim bytesRead As Integer = stream.Read(buffer, 0, buffer.Length) ' Use the initialized buffer
                 '  Dim rawResponse As String = Encoding.ASCII.GetString(buffer, 0, bytesRead) ' Convert bytes to a string
                 ' Resolver.Log.Debug($"Raw SMTP Response: {rawResponse}") ' Log the response

             End Using
         End Using

         Resolver.Log.Info("Email sent successfully.")
     Catch ex As Exception
         Resolver.Log.Error($"Failed to send email: {ex.Message}")
     End Try
 End Sub`