WildernessLabs/Meadow_Issues #578: Cloud data retention is problematic

ctacke, 2024-04-05:
the 48-hour data retention for free-tier cloud accounts is problematic because the data is not stored with a "received" timestamp.  If my device collects data but does not connect and send the data for a couple of days, the data will get deleted almost immediately upon delivery to the cloud (whatever the sweep period is on the cloud).  Additionally, if the clock on my device is incorrect (i.e. it fails to get an NTP time) then the data will have a timestamp that is before the retention limit and will similarly be deleted, which can lead the user to thinking the device isn't sending data rather than the device not having a proper clock time.

Data needs to be stored with 2 times, the "record" time and the "received" time and retention needs to be based on "received" time.

lamebrain, 2024-04-05:
@j-koda80 can you take a look at this? We have `timestamp` and `receivedAt`.

j-koda80, 2024-04-07:
@ctacke @lamebrain there is already an open issue relating to this ticket:
https://github.com/WildernessLabs/Meadow_Issues/issues/576

lamebrain, 2024-04-23:
this is fixed 
[image]

lamebrain, 2024-04-23:
this is fixed

closed 2024-04-23 by lamebrain