WildernessLabs/Meadow_Issues #581: Collections view improvements

ctacke, 2024-04-05:
The Collections view shows an ID column - no idea what that's for.  Seems like useless real estate spend.  It also doesn't show all devices in a give collection.  Yes, I can get that from the Devices view with a drop-down, but I feel there should be a way to get the list from the collections view, even if it's "click the collection and it takes you to the filtered devices view"

lamebrain, 2024-04-23:
Good suggestion. @adunndevster2 let's add a "View Devices" hover menu items that takes you to the filtered device page.

adunndevster2, 2024-04-29:
@ctacke @lamebrain - how does this look? See the video inside.

https://github.com/WildernessLabs/Meadow.Cloud.UI/pull/162