WildernessLabs/Meadow_Issues #583: State machine does not recover from Handshake failed

lamebrain, 2024-04-06:
Connection Service state machine not able to recover from `Handshake failed`.  Catch and restart? 

```
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
stdout> MQTT Error connecting to Meadow.Cloud: Handshake failed
stdout> MQTT disconnected
stdout> Connecting MQTT client
stdout> Handshake failed
```

lamebrain, 2024-04-06:
FYI this happened after the cert was updated on the MQTT host on staging cc @stevenkuhn

alexischr, 2024-06-11:
@ctacke will re-test

alexischr, 2024-06-25:
Closing, this has not been seen in a while.

closed 2024-06-25 by alexischr