WildernessLabs/Meadow_Issues #587: Device crashing after 'Too Many Open Files" message

ctacke, 2024-04-08:
I have an application (I can provide if needed, but requires specific hardware to run).  The basic premise of the app:

- Every 100ms it reads an accelerometer over I2C
- Every accelerometer change, it updates a display I2c
- Every 1s it reads power from an INA219 (I2C)
- every minute it sends the average of the current reading to Meadow.Cloud
- every 30 seconds it makes a check, with 25% chance that it will toggle to network connection state
- 
After a period of time (it's happened twice now, once after ~36 hours, once after ~3 hours) the following occurs:

- Device power draw/current spikes upward 25% or more
- Device gives a lot (> 100) of these messages when attempting to send data:

```
Debug: making cloud request to /api/events with payload: {"measurements":{"MeanCurrent":293.958533333333},"description":"Tilt Power Usage","eventId":0,"timestamp":"2024-04-08T21:24:14.8Z"}
Warning: Unable to forward Meadow Cloud record: Too many open files
```

The error message is being display when this call throws:
https://github.com/WildernessLabs/Meadow.Core/blob/95c516e53168e3d1c234c0677d7a060c33b38861/source/Meadow.Core/Cloud/MeadowCloudConnectionService.cs#L91

The error is definitely happening in the `Send` method of the `MeadowCloudConnectionService`.
The exception could be anywhere in the method, but the likely line of code throwing is `HttpClient.PostAsync`: https://github.com/WildernessLabs/Meadow.Core/blob/95c516e53168e3d1c234c0677d7a060c33b38861/source/Meadow.Core/Cloud/MeadowCloudConnectionService.cs#L690

After about 5 minutes of the above messages, I get:

```
Error: Task ID 1 faulted: One or more errors occurred. (Unable to read network information from Meadow OS)
Inner Exception: Unable to read network information from Meadow OS
Information: CRASH: Meadow will restart in 5 seconds.
```

And the device restarted. After restart, it is behaving normally again, but it did *not* appear to create a crash report file (one was not sent anyway).

halyssonJr, 2024-04-24:
@ctacke, could you share your app? I would like to test and investigate.

ctacke, 2024-04-24:
App, schematics, etc are here
https://github.com/yoshimoshi-garage/tilt

adrianstevens, 2024-05-28:
this is probably fixed

closed 2024-05-28 by adrianstevens