WildernessLabs/Meadow_Issues #589: Poll is always returning false on WiFi

duduita, 2024-04-09:
**Describe the bug**
On WiFi, the [PollRead](https://github.com/WildernessLabs/corefx/blob/f749f1129dcbcbd4e916bdaf982273782bfb98a3/src/System.Net.Http/src/System/Net/Http/SocketsHttpHandler/HttpConnection.cs#L153) is always returning false, as if the connection was always usable, leading to issues during an HTTP request, since when the connection is not usable, Mono thinks that it's.

```
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning false
Meadow StdOut: Waiting for 180000 seconds
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning false
Meadow StdOut: Waiting for 180000 seconds
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning false
Meadow StdOut: Waiting for 180000 seconds
```

On Ethernet, however, it's working as expected: it returns false when the connection is usable, and true when isn't.

```
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning false
Meadow StdOut: Waiting for 180000 seconds
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning true
Meadow StdOut: Waiting for 180000 seconds
Meadow StdOut: Polling...
Meadow StdOut: Poll is returning true
Meadow StdOut: Waiting for 180000 seconds
```

**To Reproduce**
Just run the [PollBugReproApp.txt](https://github.com/WildernessLabs/Meadow_Issues/files/14919463/PollBugReproApp.txt) sample app using WiFi and then using Ethernet to see the difference.

**Expected behavior**
WiFi should return false in the first poll, and after a few minutes, it should return true, meaning that the connection is not usable anymore.

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
```
Board Information 
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: MeadowCoreCompute

Hardware Information 
    Processor type: STM32F777IIK6
    ID: 2F-00-1F-00-12-50-33-53-32-34-31-20
    Serial number: 2050346B5333
    Coprocessor type: ESP32
    MAC Address - 
        WiFi: 4C:75:25:D5:28:5C

Firmware Versions 
    OS: 1.10.0.2
    Mono: 1.10.0.2
    Coprocessor: 1.10.0.1
    Protocol: 7
```

alexischr, 2024-04-30:
@duduita is this differnet from https://github.com/WildernessLabs/Meadow/commit/3c480575e3895d3bb237b9daf4a5c4f413733018 ?

duduita, 2024-04-30:
No, this commit that you mentioned (PR [#587](https://github.com/WildernessLabs/Meadow/pull/587)) aims to fix this issue current (#589), so we can close it after releasing 1.11.0.0 @alexischr.

closed 2024-05-14 by alexischr