WildernessLabs/Meadow_Issues #590: Socket.Send throw exception with 'The descriptor is not a socket'

ctacke, 2024-04-09:
Meadow F7 Project Lab
OS v 1.10.0.2

[MeadowSocketSend Test application attached](https://github.com/WildernessLabs/Meadow_Issues/files/14920569/MeadowSocketSend.zip)

I have a pair of very simple applications, a client and sender that use a Socket for communication.
The "Client" app runs on a PC, accepts and listens for a sender to attach.
The "Sender" connects, and sends a simple text message.

This works fine on a desktop machine.
On the Meadow, the call to `Send` throw an exception with the message "The descriptor is not a socket"

jbeenes, 2024-04-09:
We have been struggling with WebSockets on Meadow too, using System.Net.WebSockets. Did you test with this aswell?

In our search to get our test working I came across this [link](https://learn.microsoft.com/en-us/dotnet/api/system.net.websockets.clientwebsocket?view=net-8.0), where the part under Remarks made me think this might never work on something other then Windows. Do you agree?

We made it work eventually using a library called [SocketSharp](https://github.com/sta/websocket-sharp).

djaus2, 2024-04-10:
I am currently experiencing a related problem **"The descriptor is not a socket"**.

Situation:
- **TCP Service running on an Arduino device**
-  **Implemented as a .NET Class library** 
  - .NET 6 or 8 and .NET Standard 2.1
  - Client = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);
    - Also tried as TCPClient with same result.
- All works OK from desktop as 2 apps
  - .NET Console app with .NET Class
  - Blazor App with .NET Class 
- In Meadow context ProjectLabs device
  - Using .NET Standard version of Library 
  - Can connect
  - Get "The descriptor is not a socket" error when sending to service
  - Of note is that the Client.Available property returns that error.
- Can run OK as Meadow.Windows app on desktop using either .NET or .NET Standard class library.

Have drilled into this a bit on Slack: https://wildernesslabspublic.slack.com/archives/CB9BW8HQE/p1712460726231349

duduita, 2024-04-22:
The core of this issue lies in the `Socket.Available` property, which relies on a native function that is currently unsupported by our operating system version. The OS rebase, expected in the coming months, should definitively resolve this problem.

As a temporary solution, the `Socket.Available` property will return `1` when there are bytes ready to be read and `0` when there aren't. While this workaround may not provide exact data availability, it serves as a reliable indicator of whether the socket contains readable data.

djaus2, 2024-04-23:
Thanks for the update.
David Jones

ctacke, 2024-05-01:
Fixed in 1.11

closed 2024-05-01 by ctacke