WildernessLabs/Meadow_Issues #591: Meadow.OS - Consolidate NetworkDisconnected event and IsConnected property

jorgedevs, 2024-04-09:
Currently, `NetworkDisconnected` events and `IsConnected` property behave differently if you're working with network via config files vs using just C#.

- With Config files: `NetworkDisconnected` only fires after running out of tries to join a network (in intervals of 30s approx), so If i turn off my network, it'll take the number of tries x 30s to finally fire that event, so all that time, my app will appear that I'm still online which im not, so even checking `IsConnected` before doing API calls will fail.
- With C#, since there's not reconnect attempts mechanisms, when I turn off my network, `NetworkDisconnected` fires immediately and `IsConnected` is false right away.

Expected behavior

`NetworkDisconnected` event and `IsConnected` property must behave the same for both cases. When the ESP detects that it lost network connection, the event and property flag should change immediately but also fire up the reconnect mechanism. We could add a new event that triggers when it ran out of reconnect attempts.

ctacke, 2024-04-09:
I'd advocate for all mechanisms raise "Connected" when they actually connect, "Disconnected" when they actully disconnect, and for the auto-connect add something like "AutoConnectAbandoned" for when the auto-connect has cycled through all of its attempts and given up

doingnz, 2024-04-09:
Please consider the steps the app / Meadow OS need to take to scan for, poll, ??? to reconnect to a network after the disconnected event. 

Connect/disconnect should be raised when they happen given only end app developer knows what they want their solution to do when the wifi is gone/out of range. (Log data to SQlite, discard, etc... )

Consider an IoT device in a truck that connects & disconnects to free hotspots/depot when they become available to upload data / logs vs a stationary Clima in the bottom of the garden that may only get intermittent wifi failures or interference from neighbours.

halyssonJr, 2024-04-11:
@jorgedevs and @ctacke, it seems the coprocessor reconnects to AP but doesn't fire the connect event, this way  `IsConnected` never updates from **false** to **true, causing a different behavior between C# and  Config file. @jorgedevs could you share a minimal reproduce this behavior? Looking at the ESP32 logs the reconnection retry doesn't take 30 seconds. Check this code, Does it look closer to the scenario?

```
using Meadow;
using Meadow.Devices;
using Meadow.Hardware;
using Meadow.Foundation;
using System;

namespace MeadowApp
{
    public class MeadowApp : App<F7CoreComputeV2>
    {
        private IWiFiNetworkAdapter wifi;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.NetworkConnected += Wifi_NetworkConnected;
            wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
            wifi.NetworkError += Wifi_NetworkError;
            return base.Initialize();
        }

        public override async Task Run()
        {
            Console.WriteLine("Run...");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == false
            Console.WriteLine("IPAddress = " + wifi.IpAddress);
            while(true)
            {
                if (wifi.IsConnected == false)
                {
                    Console.WriteLine("Wait to reconnect...");
                }
                else
                {
                    Console.WriteLine("Connected");

                }
                await Task.Delay(1000);
            }
        }

        private void Wifi_NetworkError(INetworkAdapter sender, NetworkErrorEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkError event fired...");
        }

        private void Wifi_NetworkDisconnected(INetworkAdapter sender, NetworkDisconnectionEventArgs args)
        {
            Console.WriteLine($"wifi.IsConnected (from Wifi_NetworkDisconnected event) = {wifi.IsConnected}");
        }

        private void Wifi_NetworkConnected(INetworkAdapter sender, NetworkConnectionEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkConnected event fired...");
        }
    }
}
```

alexischr, 2024-04-22:
After a fruitful [discussion on private Slack](https://wildernesslabs.slack.com/archives/C023KR4Q8QJ/p1713199287732339), a diagram was generated by @ctacke, attached below. This is guiding us toward the right changes to our network event model.

Source: https://gist.github.com/alexischr/e4096a2299c4a5329d16f66e65325bc5

[image]

doingnz, 2024-04-22:
Nice diagram!

Does raise "connect failed" give a reason? i.e. differentiate between wrong SSID/password vs wifi is unreachable?

Starting the wifi connect at 'device start' in the ESP32 is a great way to shorten the connect time on boot. Just consider & document how the application should know if & when that connection was successful. i.e. how to cope with the race condition where "connected" event might be raised before or after the Meadow `Initialize()` is called and the app has had an opportunity to register an event handler. The app may want to perform its own actions when the connected event is raised and depends on seeing the event. One idea might be to raise the connected event after `Initialise()` and before `Run()` when connection started from `device start`. That gives the app a chance to register event handler. Or it could be the App's responsibility to check `IsConnected` during `Initialize()` and call it's event handler. There may be other approaches. Just need to say what is expected. 

Please explain how NTP fits into this diagram. Data logging applications may not wish to run if date or time in question. Will auto NTP be triggered when `connected` raised every time? similar race condition about NTP time changed event. Time may still be valid if wifi disconnect happened briefly after previous successful connect. Agin document what support from Meadow and what the app is expected to do.

doingnz, 2024-05-26:
@ctacke Great to see this is fixed in an upcoming release. I assume the changes are already part of develop branch.

Is there a sample application for testing the above flow chart?

closed 2024-05-28 by adrianstevens

doingnz, 2024-05-28:
Curious about issues getting closed with open questions attached?

Closed #591<https://github.com/WildernessLabs/Meadow_Issues/issues/591> as completed.