WildernessLabs/Meadow_Issues #597: MicroJson - Struggling to Deserialize date time in a certain format

jorgedevs, 2024-04-16:
Running my Meadow.Cloud logging sample on Project Lab to log a few events, I then use the Meadow.Cloud API sample to get those records and graph them on the screen, im getting an issue deserializing the json response.

Here's a sample:

```json
{
    "data": {
        "totalRecords": 131,
        "queryTimeInMilliseconds": 207,
        "queryResponses": [
            {
                "id": "-Jm56I4BqSlQUq-Q5c1L",
                "record": {
                    "eventId": 1000,
                    "description": "environment reading",
                    "receivedAt": "2024-04-16T21:04:40.6806972Z",
                    "deviceId": "23-00-3D-00-15-50-33-4D-35-34-33-20",
                    "orgId": "d82f8ec7243a461581c2544070209fbd",
                    "measurements": {
                        "temperature": "26.65",
                        "humidity": "33.70",
                        "pressure":"1,1427.81"
                    },
                    "timestamp": "2024-04-16T21:04:39.181Z"
                }
            },
            {
                "id": "2Hq46I4BSI8lMR6P6-Ua",
                "record": {
                    "eventId": 1000,
                    "description": "environment reading",
                    "receivedAt": "2024-04-16T21:03:36.628572Z",
                    "deviceId": "23-00-3D-00-15-50-33-4D-35-34-33-20",
                    "orgId": "d82f8ec7243a461581c2544070209fbd",
                    "measurements": {
                        "temperature": "26.66",
                        "humidity": "33.78",
                        "pressure": "1,027.83"
                    },
                    "timestamp": "2024-04-16T21:0335.148Z"
                }
            },
            {
                "id": "03q36I4BSI8lMR6P8eUW",
                "record": {
                    "eventId": 1000,
                    "description": "environment reading",
                    "receivedAt": "2024-04-16T21:02:32.5948251Z",
                    "deviceId": "23-00-3D-00-15-50-33-4D-35-34-33-20",
                    "orgId": "d82f8ec7243a461581c2544070209fbd",
                    "measurements": {
                        "temperature": "26.73",
                        "humidity": "33.92",
                        "pressure": "1,027.89"
                    },
                    "timestamp": "2024-04-16T21:02:31.107Z"
                }
            }
        ]
    },
    "isSuccessful": true,
    "erorMessage": null
}
```

Using that sample in the MicroJson M.F. sample (importing the required model classes). I'm getting the following error:

```console 
String '2024-04-16T21:0335.148Z' was not recognized as a valid DateTime.
```

adrianstevens, 2024-04-16:
As discussed, the timestamp isn't a valid date time so there's an encoding issue

adrianstevens, 2024-04-16:
Renaming issue

jorgedevs, 2024-04-16:
@adrianstevens we found the issue, my sample app had an `object` field that had to be a `string?`, so Meadow.Cloud and MicroJson are good, was just a bad DTO, apologies.

closed 2024-04-16 by jorgedevs