WildernessLabs/Meadow_Issues #599: Bmx280 Exception on SPI, zero value on I2C

TheFlyingDude, 2024-04-17:
**Describe the bug**
I can't get any meaningfull data from the BME280 sensors.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup BME280 sensor using the example from https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme280/
2. Wire it up for SPI or I2C
3. Try initializing new BME280 object in SPI mode -> ArgumentOutOfRangeException
4. Try reading data in I2C mode -> all data values come back as 0

**Expected behavior**
Working communication with sensor in both SPI and I2C mode

**Developer tools**
 - Windows 11 23H2 
 - IDE and version: Microsoft Visual Studio Community 2022 (64-bit) Version 17.9.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.9.7

**Meadow**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV1 (Slack user reported having the same problems with the F7v2)
 - Meadow OS version: 1.10.0.2
 - Meadow coprocessor version: 1.10.0.1

**Additional context**
- GY-BME280 breakout board 

I tried both SPI and I2C mode with the respective wiring (with and without the 2x 4.7kOhm pullup resistors). In I2C it seems like the sensor gets found under the default address, but only returns 0 for temp. humidity and pressure.
When i try to instantiate a new BME280 in SPI mode, it crashes with following error:

Create BME280 sensor with SPI...
System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
at Meadow.Foundation.SpiCommunications.ReadRegister (System.Byte address, System.Span`1[T] readBuffer) [0x00055] in <c097d0c14b5a44a8a58c7a3bb2920c69>:0
at Meadow.Foundation.Sensors.Atmospheric.Bmx280.ReadCompensationData (Meadow.Hardware.IByteCommunications bmx280Comms, System.Memory
1[T] readBuffer, Meadow.Foundation.Sensors.Atmospheric.CompensationData compensationData) [0x00018] in <10d76579efa147a787caa700ca1ce042>:0
at Meadow.Foundation.Sensors.Atmospheric.Bme280.Initialize () [0x00000] in <10d76579efa147a787caa700ca1ce042>:0
at Meadow.Foundation.Sensors.Atmospheric.Bme280..ctor (Meadow.Hardware.ISpiBus spiBus, Meadow.Hardware.IDigitalOutputPort chi
SelectPort) [0x0006e] in <10d76579efa147a787caa700ca1ce042>:0
at Meadow.Foundation.Sensors.Atmospheric.Bme280..ctor (Meadow.Hardware.ISpiBus spiBus, Meadow.Hardware.IPin chipSelectPin) [0x00009] in <10d76579efa147a787caa700ca1ce042>:0
at MeadowTest2.MeadowApp.CreateSpiSensor () [0x0004b] in <7a38836d163d46618e2b3aeb75b21ac0>:0
at MeadowTest2.MeadowApp.Initialize () [0x000
3] in <7a38836d163d46618e2b3aeb75b21ac0>:0

**Nuget package**
https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme280/

JT-50, 2024-04-24:
this problem was fixed here: https://github.com/WildernessLabs/Meadow.Foundation/pull/953

ctacke, 2024-05-01:
Fixed in 1.11

closed 2024-05-01 by ctacke