WildernessLabs/Meadow_Issues #603: Meadow Cloud should show "provisioned" date

ctacke, 2024-04-18:
I have a cloud account with a dozen devices.  When I provision a new F7 device it shows up with the name "MeadowF7", but I already have a few of these so I can't tell which one was just added.  It would be great if it has a column for "date provisioned" and maybe even a "new!" flag by devices provisioned within the last hour or whatever.

lamebrain, 2024-04-23:
@ctacke could a workaround be to use the `-n` flag to specify a unique name for the added device? My only hesitation is that space is limited in the grid.

adunndevster2, 2024-04-24:
There could be room for another column:
[image]

That said, at some point we may want to make use of more sophisticated grid features, liking putting in a custom component on each row that show's data however we want...

{"type":"excalidraw/clipboard","elements":[{"type":"rectangle","version":125,"versionNonce":1946406580,"index":"bG1","isDeleted":false,"id":"Cu1fTCqTH2cUIcCuNsGea","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4261.541148827327,"y":1179.2070591037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":498.6484375,"height":28.1875,"seed":675170316,"groupIds":[],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971653860,"link":null,"locked":false},{"type":"text","version":8,"versionNonce":401490700,"index":"bG2","isDeleted":false,"id":"My9o9_5C456sechH3G6XR","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4278.474742577327,"y":1181.8867466037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":22.279983520507812,"height":25,"seed":892335372,"groupIds":[],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971668727,"link":null,"locked":false,"fontSize":20,"fontFamily":1,"text":"Id","textAlign":"left","verticalAlign":"top","containerId":null,"originalText":"Id","lineHeight":1.25},{"type":"text","version":130,"versionNonce":1219632012,"index":"bG3","isDeleted":false,"id":"vOEmyOcSo-UKWte6EJdtD","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4355.377719567073,"y":1181.7500278537957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":49.81996154785156,"height":25,"seed":917106700,"groupIds":[],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971677751,"link":null,"locked":false,"fontSize":20,"fontFamily":1,"text":"Name","textAlign":"left","verticalAlign":"top","containerId":null,"originalText":"Name","lineHeight":1.25},{"type":"text","version":370,"versionNonce":2049652492,"index":"bG4","isDeleted":false,"id":"DguOXI-G21-5j6iqkp132","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4462.420230553401,"y":1180.9922153537957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":120.91989135742188,"height":25,"seed":2047079988,"groupIds":[],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971692148,"link":null,"locked":false,"fontSize":20,"fontFamily":1,"text":"Extra stuff","textAlign":"left","verticalAlign":"top","containerId":null,"originalText":"Extra stuff","lineHeight":1.25},{"type":"rectangle","version":333,"versionNonce":1824262580,"index":"bG4V","isDeleted":false,"id":"6Hgq-wVOnxTU9Vk70ZfPF","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4261.607555077327,"y":1207.3984653537957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":497.3671875,"height":115.06640625,"seed":1309162164,"groupIds":["hoe5h-eEy5ef033FPGHMV"],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971799511,"link":null,"locked":false},{"type":"line","version":122,"versionNonce":1801814964,"index":"bG5","isDeleted":false,"id":"BAW5ASXb89qWqFfg9G3pF","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4319.638805077327,"y":1208.3945591037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":0,"height":112.890625,"seed":1631501364,"groupIds":["hoe5h-eEy5ef033FPGHMV"],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971799510,"link":null,"locked":false,"startBinding":null,"endBinding":null,"lastCommittedPoint":null,"startArrowhead":null,"endArrowhead":null,"points":[[0,0],[0,112.890625]]},{"type":"line","version":135,"versionNonce":174899596,"index":"bG6","isDeleted":false,"id":"-GCaHuLQJvBrx9UnMuneb","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4430.994776868908,"y":1206.7539341037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":0,"height":112.890625,"seed":160021684,"groupIds":["hoe5h-eEy5ef033FPGHMV"],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971799510,"link":null,"locked":false,"startBinding":null,"endBinding":null,"lastCommittedPoint":null,"startArrowhead":null,"endArrowhead":null,"points":[[0,0],[0,112.890625]]},{"type":"text","version":67,"versionNonce":820178228,"index":"bG7","isDeleted":false,"id":"DJWk7quBE4zFpLiLqGSt1","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4279.064586327327,"y":1251.4375278537957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":16.439987182617188,"height":25,"seed":221088780,"groupIds":["hoe5h-eEy5ef033FPGHMV"],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971799510,"link":null,"locked":false,"fontSize":20,"fontFamily":1,"text":"...","textAlign":"left","verticalAlign":"top","containerId":null,"originalText":"...","lineHeight":1.25},{"type":"text","version":208,"versionNonce":1962192908,"index":"bG8","isDeleted":false,"id":"nn5JwsRADQK93Zo2VJ6W-","fillStyle":"solid","strokeWidth":2,"strokeStyle":"solid","roughness":1,"opacity":100,"angle":0,"x":4368.621936486018,"y":1251.5820591037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":16.439987182617188,"height":25,"seed":1631760012,"groupIds":["hoe5h-eEy5ef033FPGHMV"],"fr [... cut ...] ","roughness":1,"opacity":100,"angle":0,"x":4613.342311547053,"y":1480.6211216037957,"strokeColor":"#1e1e1e","backgroundColor":"transparent","width":63.55995178222656,"height":25,"seed":1296491148,"groupIds":["FLxBwJmsXKwER2UelR2fP"],"frameId":null,"roundness":null,"boundElements":[],"updated":1713971811476,"link":null,"locked":false,"fontSize":20,"fontFamily":1,"text":"etc: ...","textAlign":"left","verticalAlign":"top","containerId":null,"originalText":"etc: ...","lineHeight":1.25}],"files":{}}