WildernessLabs/Meadow_Issues #605: BME280 driver is non-op

ctacke, 2024-04-20:
Trying to create a VB sample for the BME280 and I see the driver is returning all zeros for data:

```
Public Overrides Async Function Run() As Task
    Dim conditions = Await sensor.Read()
    Resolver.Log.Info("Initial Readings:")
    Resolver.Log.Info($"  Temperature: {conditions.Temperature?.Celsius:N2}C")
    Resolver.Log.Info($"  Pressure: {conditions.Pressure?.Bar:N2}hPa")
    Resolver.Log.Info($"  Relative Humidity: {conditions.Humidity?.Percent:N2}%")
End Function
```
yields
```
Temperature: 0.00C
Pressure: 0.00hPa
Relative Humidity: 0.00%
```

I instrumented the Bmx280 code for reading compensation and sensor data to output the raw register read values, and I get non-zero data (and changing data in the values register)
for example, this code:
```
        Resolver.Log.Info("Reading 0x88...");
        // read the temperature and pressure data into the internal read buffer
        bmx280Comms.ReadRegister(0x88, readBuffer.Span[0..24]);
        Resolver.Log.Info($"Read {BitConverter.ToString(readBuffer.ToArray())}");
```
Yields output like this
```
Reading 0x88...
Read B1-40-5C-67-03-F0-35-8A-A7-D7-58-00-9B-12-20-00-12-1E-00-00-D9-00-46-F1-00-00-00-00-00-00-00-00
Reading 0xf7...
Read 62-93-80-78-17-80-46-D6-A7-D7-58-00-9B-12-20-00-12-1E-00-00-D9-00-46-F1-00-00-00-00-00-00-00-00
```
I then modified `ReadSensor` to see what the calculations are doing:
```
        Resolver.Log.Info("Reading 0xf7...");
        bmx280Comms.ReadRegister(0xf7, readBuffer.Span[0..8]);
        Resolver.Log.Info($"Read {BitConverter.ToString(readBuffer.ToArray())}");

        var adcTemperature = (readBuffer.Span[3] << 12) | (readBuffer.Span[4] << 4) | ((readBuffer.Span[5] >> 4) & 0x0f);
        var tvar1 = (((adcTemperature >> 3) - (compensationData.T1 << 1)) * compensationData.T2) >> 11;
        var tvar2 = (((((adcTemperature >> 4) - compensationData.T1) *
                       ((adcTemperature >> 4) - compensationData.T1)) >> 12) * compensationData.T3) >> 14;
        var tfine = tvar1 + tvar2;
        var tcalc = (((tfine * 5) + 128) >> 8) / 100;

        Resolver.Log.Info($"Read {adcTemperature} {tvar1} {tvar2} {tfine} {tcalc}");
        conditions.Temperature = new Units.Temperature(tcalc, TU.Celsius);
```
And I get this:
```
Read 491896 0 0 0 0
```
It feels like all of the compensation data values are zero and therefore zeroing out the calculation

ctacke, 2024-04-20:
Did some work on debugging this.  The first problem was compensation data is a struct but passed by value, so it's always zero.  I fixed that and now I get non-zero data back, but the temp is obviously wrong (returns ~69.1C in my office).  The math seems to match the Philips docs and the Adrduino driver, and both yield the same end result of an obviously wrong value.  

Tried 2 different parts and the results are the same errorneous temp, so it's not bad hardware, just mathing bad or reading wrong register somewhere.

adrianstevens, 2024-04-20:
Fixed here https://github.com/WildernessLabs/Meadow.Foundation/pull/953

jorgedevs, 2024-04-23:
PR Fix its merged, so I'll mark it as "fixed in upcoming release"

ctacke, 2024-05-01:
Fixed in 1.11

closed 2024-05-01 by ctacke