WildernessLabs/Meadow_Issues #606: Cultivar silently stop sending cloud data without crashing the app

duduita, 2024-04-22:
**Describe the bug**
It pushed logs to the cloud for ~1200 events, after that the app still is responding to button interrupts, and is still collecting sensor data, but the output looks like this:

```
stdout> Temperature: 30.6 | Humidity: 27.9 | Moisture: 2.3
stdout> making cloud request to /api/events with payload: {"measurements":{"TemperatureCelsius":30.630859375,"HumidityPercent":27.9277185503018,"SoilMoistureDouble":2.32912550513668},"description":"Atmospheric reading","eventId":110,"timestamp":"2024-04-17T21:59:30.054Z"}
stdout> cloud request to /api/events completed successfully
stdout> Temperature: 30.5 | Humidity: 28.9 | Moisture: 2.3
stdout> making cloud request to /api/events with payload: {"measurements":{"TemperatureCelsius":30.515625,"HumidityPercent":28.9144800926825,"SoilMoistureDouble":2.3297915058473},"description":"Atmospheric reading","eventId":110,"timestamp":"2024-04-17T22:01:30.292Z"}
stdout> Temperature: 30.7 | Humidity: 27.5 | Moisture: 2.3
stdout> Temperature: 30.7 | Humidity: 28.7 | Moisture: 2.3
stdout> Temperature: 30.7 | Humidity: 27.7 | Moisture: 2.3
stdout> Temperature: 30.8 | Humidity: 27.0 | Moisture: 2.3
stdout> Temperature: 30.8 | Humidity: 27.0 | Moisture: 2.3
stdout> Temperature: 30.9 | Humidity: 27.3 | Moisture: 2.3
stdout> Temperature: 30.9 | Humidity: 27.4 | Moisture: 2.3
stdout> Temperature: 30.9 | Humidity: 26.2 | Moisture: 2.3
```
So it's stopped sending cloud data. I see it did a making cloud request and then never reported error or success, and just quit reporting anything at all.

Additionally, after enabling mbedTLS native logs, I could see that it stopped during the flush output:

```
Meadow StdOut: ssl_msg.c:2016: => flush output
Meadow StdOut: ssl_msg.c:2034: message length: %zu, out_left: %zu
Meadow StdOut: Temperature: 37.5 | Humidity: 33.4 | Moisture: 2.3
Meadow StdOut: Temperature: 37.5 | Humidity: 33.4 | Moisture: 2.3
Meadow StdOut: Temperature: 37.5 | Humidity: 33.4 | Moisture: 2.3
Meadow StdOut: Temperature: 37.5 | Humidity: 33.6 | Moisture: 2.3
...
```

**To Reproduce**
Run CultivarApp with health metrics enabled.

**Expected behavior**
It should show an error, instead of silently fail.

alexischr, 2024-04-30:
We need to split this issue in two:

* A network deadlock is affecting our reliability
* The deadlock is not detected, so the device never attempts to reset/recover

duduita, 2024-05-03:
Created a separate [issue](https://github.com/WildernessLabs/Meadow_Issues/issues/622) focusing on make the device reset/recover when a deadlock is detected.

bryancostanich, 2024-08-06:
Believe that this might be fixed. @duduita to confirm.

bryancostanich, 2024-09-05:
Probably caused by the deadlock which has been fixed.

closed 2024-09-05 by bryancostanich