WildernessLabs/Meadow_Issues #610: Cultivar app causing OOM errors with cellular network

duduita, 2024-04-26:
**Describe the bug**
I'm getting out-of-memory errors when running Cultivar with cellular network.

```
Meadow StdOut: [0x1e: 252.41400 11] LEAVE:c System.Net.Sockets.NetworkStream:ge
Meadow StdOut: t_CanRead ()(result=1
Meadow StdOut: [0x1e: 252.41600 11] ENTER:c System.Net.Sockets.NetworkStream:ge
Meadow StdOut: t_CanWrite ()(this:0xc03b5c98[System.Net.Sockets.NetworkStream M
Meadow StdOut: eadow.dll])
Meadow StdOut: [0x1e: 252.41800 11] LEAVE:c System.Net.Sockets.NetworkStream:ge
Meadow StdOut: t_CanWrite ()(result=1
Meadow StdOut: memalign returned null when requesting 1048576 bytes (1048576 alignment) -- out of memory?
Meadow StdErr: Unrecoverable .NET Runtime error. Meadow will restart in 5 secon
Unrecoverable .NET Runtime error. Meadow will restart in 5 secon
```

```
Meadow StdOut: [0x1b: 274.75000 16] ENTER:c Mono.MbedTls.MbedTlsContext:get_IsA
Meadow StdOut: uthenticated ()(this:0xc0396320[Mono.MbedTls.MbedTlsContext Mead
Meadow StdOut: ow.dll])
Meadow StdOut: [0x1b: 274.75300 16] LEAVE:c Mono.MbedTls.MbedTlsContext:get_IsA
Meadow StdOut: uthenticated ()(result=1
Meadow StdOut: [0x1b: 274.75500 16] ENTER:c Mono.MbedTls.MbedTlsContext:Read (b
Meadow StdOut: yte[],int,int)(this:0xc0396320[Mono.MbedTls.MbedTlsContext Meado
Meadow StdOut: w.dll], [System.Byte[]:[4096]0xc03ab5e0], 0, 4096)
Meadow StdOut: [0x1b: 274.75700 17] ENTER:c (wrapper managed-to-native) Mono.Mb
Meadow StdOut: edTls.MbedTlsContext:mono_mbedtls_read (intptr,int)(0xc1b6ddf0,
Meadow StdOut: 4096)
Meadow StdOut: [0x1b: 274.89400 17] LEAVE:c (wrapper managed-to-native) Mono.Mb
Meadow StdOut: edTls.MbedTlsContext:mono_mbedtls_read (intptr,int)(result=143
Meadow StdOut: [0x1b: 274.89600 16] LEAVE:c Mono.MbedTls.MbedTlsContext:Read (b
Meadow StdOut: yte[],int,int)((unknown return type 15)
Meadow StdOut: memalign returned null when requesting 1048576 bytes (1048576 alignment) -- out of memory?
Meadow StdErr: Unrecoverable .NET Runtime error. Meadow will restart in 5 secon
Unrecoverable .NET Runtime error. Meadow will restart in 5 secon

```
**To Reproduce**
Run CultivarApp with health metrics enabled (1 minute interval), using cellular network.

**Expected behavior**
As it works on WiFi and Ethernet, it should be able to run for long hours normally.

**Additional context**
Using cellular with other apps, I can't see the OOM errors, It's the first app that I see cellular causing OOM errors.

**Meadow (please complete the following information as best as you can):**
Board Information
Model: F7Micro
Hardware version: F7CoreComputeV2
Device name: Cultivar

Hardware Information
Processor type: STM32F777IIK6
ID: 2F-00-1F-00-12-50-33-53-32-34-31-20
Serial number: 2050346B5333
Coprocessor type: ESP32
MAC Address -
WiFi: 4C:75:25:D5:28:5C

Firmware Versions
OS: 1.11.0.0
Mono: 1.11.0.0
Coprocessor: 1.11.0.0
Protocol: 7

jorgedevs, 2024-07-04:
Running Cultivar, I'm also seeing random silent resets, so looks like it happens with all network adapters.

bryancostanich, 2024-08-06:
Likely NTP client. NTP has now been moved out of Kernel and into Userland, so we're hopeful that will fix this.

bryancostanich, 2024-08-13:
Working on getting the NTP PR merged. Lots of work to review and make sure we get this right. That should fix this issue.

bryancostanich, 2024-08-27:
We're no longer confident that this is NTP. Investigating.

duduita, 2024-09-22:
After seeing Cultivar running for 25 days without silently and randomly restarting, this issue looks fixed to me.

I should open a separate issue to improve our ability to report errors, such as sending to the cloud OOM, deadlock, and other types of errors that are currently not being sent.

closed 2024-09-22 by bryancostanich

NevynUK, 2024-11-18:
Marking as Done as the original issue is now closed..