WildernessLabs/Meadow_Issues #620: ignore mpak folder on deploy

lamebrain, 2024-05-03:
expected:
mpak folder contents are not deployed

actual
mpak folder contents are deployed

```
Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024031634.mpak'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024031845.mpak'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024032312.mpak'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024032125.mpak'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024041683.zip'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024031572.mpak'

Sending  '/Users/briankim/gh/OtaSample/bin/Release/netstandard2.1/mpak/2024041683/info.json'
```

adrianstevens, 2024-05-03:
ah yes :) let's fix that

adrianstevens, 2024-05-03:
Fixed in this PR https://github.com/WildernessLabs/Meadow.CLI/pull/558

adrianstevens, 2024-05-03:
CLI v2.0.39 is live - closing issue

closed 2024-05-03 by adrianstevens