WildernessLabs/Meadow_Issues #625: MicroLayout - `IsVisible` on creating an AbsoluteLayout does not work

jorgedevs, 2024-05-06:
Using MicroLayout [sample](https://github.com/WildernessLabs/Meadow.Foundation/tree/develop/Source/Meadow.Foundation.Libraries_and_Frameworks/Graphics.MicroLayout/Samples/HMI_Sample), consider the snippet:

```csharp
var chartLayout = new AbsoluteLayout(screen, 0, 0, screen.Width, screen.Height)
{ 
    IsVisible = false
};
```

**Expected Behavior** 
Whatever controls im adding to this "hidden" `chartLayout` wont be drawn on the screen.

**Actual Behavior**
Layout is not hidden and it'll keep updating showing all the controls as they're being added.

**Workaround**
Once finished loading all the controls to `chartLayout`, you can hide the entire screen doing `chartLayout.IsVisible = false;`

adrianstevens, 2025-01-12:
Fixed in develop - will ship in the next release

adrianstevens, 2025-01-21:
Fixed in v2.0.1 - closing issue

closed 2025-01-21 by adrianstevens