WildernessLabs/Meadow_Issues #626: Networking doesn't work after sleep/wake

lamebrain, 2024-05-06:
Currently Sleep works as expected in simple cases without networking. The app below sleeps for 5s and runs for 5s, continuously.
```
public class MeadowApp : App<F7FeatherV2>
{
    RgbPwmLed onboardLed;

    public override async Task Run()
    {
        Device.PlatformOS.AfterWake += async (sender, source) =>
        {
            onboardLed.SetBrightness(1);
            await Task.Delay(5000);
            Device.PlatformOS.Sleep(TimeSpan.FromSeconds(5));
        };

        Device.PlatformOS.BeforeSleep += () =>
        {
            onboardLed.SetBrightness(0);
        };
        
        onboardLed.SetColor(Color.Red, 0);
        Device.PlatformOS.Sleep(TimeSpan.FromSeconds(5));
    }

    public override Task Initialize()
    {
        onboardLed = new RgbPwmLed(
            redPwmPin: Device.Pins.OnboardLedRed,
            greenPwmPin: Device.Pins.OnboardLedGreen,
            bluePwmPin: Device.Pins.OnboardLedBlue,
            CommonType.CommonAnode);
            
        return base.Initialize();
    }
}
```

However, after waking up, there is a hang when trying to send a message.
```
public class MeadowApp : App<F7FeatherV2>
{
    RgbPwmLed onboardLed;

    public override async Task Run()
    {
        var cloudLogger = new CloudLogger();
        Resolver.Log.AddProvider(cloudLogger);

        var mcs = Resolver.Services.Get<IMeadowCloudService>();
        mcs.ConnectionStateChanged += (sender, state) =>
        {
            if (state == CloudConnectionState.Connected)
            {
                Device.PlatformOS.Sleep(TimeSpan.FromSeconds(60));
            }
        };
        
        Device.PlatformOS.BeforeSleep += () =>
        {
            onboardLed.SetColor(Color.Green, 1);
            
            // go to sleep after all queued messages have been sent
            // APP HANGS HERE AFTER THE FIRST WAKE UP

            while (mcs.QueueCount > 0)
            {
                Thread.Sleep(2000);
                Resolver.Log.Trace($"queue count: {mcs.QueueCount}");
            }
            onboardLed.SetColor(Color.Red, 0);
        };
        
        Device.PlatformOS.AfterWake += async (sender, source) =>
        {
            onboardLed.SetBrightness(1);
            Resolver.Log.Info("queue message after wake up.");
            await Task.Delay(2000);
            Device.PlatformOS.Sleep(TimeSpan.FromSeconds(60));
        };
        
        onboardLed.SetColor(Color.Red, 1);

        Resolver.Log.Info("queue up message to be sent.");
        Resolver.Log.Info("queue another message to be sent.");
    }

    public override Task Initialize()
    {
        onboardLed = new RgbPwmLed(
            redPwmPin: Device.Pins.OnboardLedRed,
            greenPwmPin: Device.Pins.OnboardLedGreen,
            bluePwmPin: Device.Pins.OnboardLedBlue,
            CommonType.CommonAnode);
            
        return base.Initialize();
    }
}
```

NevynUK, 2024-05-07:
Duplicate of this issue?

https://github.com/WildernessLabs/Meadow/issues/265.

halyssonJr, 2024-12-18:
This issue was fixed. 
### Setup
- OS: 2.0.0.3
- Runtime: 2.0.0.3
- Coprocessor: 2.0.0.0

Logs:
[logs_wakeup.txt](https://github.com/user-attachments/files/18183890/logs_wakeup.txt)

NevynUK, 2024-12-18:
How long did you sleep for?  Have you tried this with a sleep period of more than 30 minutes?

halyssonJr, 2024-12-19:
Actually, after you commented about that, I tested it, and it works fine. Should I test it for long hours (2h, 5h, 10h)?

```
[cloud] making cloud request to /api/events with payload: {"measurements":{"health.cpu_temp_celsius":47.5347985347985,"health.memory_used":1294256,"health.disk_space_used":22182779,"info.os_version":"2.0.0.3","up_time":"00:01:21.9920000","health.battery_percentage":0,"health.total_memory_arena":30457856,"health.total_memory_free_blocks":42,"health.largest_memory_free_block":924620
,"health.total_allocated_memory":21164768,"health.total_free_memory":9293088,"info.coprocessor_os_version":"2.0.0.0"},"description":"device.health","eventId":10,"timestamp":"2024-12-19T11:52:19.6032080+00:00"}
[application] queue count: 1
[cloud] cloud request to /api/events completed successfully
[application] queue count: 0
[application] Sleeping...
[cloud] making cloud request to /api/logs with payload: {"timestamp":"2024-12-19T12:23:02.0980000Z","severity":"Information","message":"[application] queue message after wake up.","exception":null}
```

NevynUK, 2024-12-19:
I would probably do one long test, say 12 hours and see how that goes.

closed 2025-01-07 by ctacke