WildernessLabs/Meadow_Issues #638: CLI 2.0.41 does not trim apps

alexischr, 2024-05-16:
**Describe the bug**
CLI deploys full-size assemblies without trimming, or a warning that linking didn’t work, etc. No pre/postlink_bin  directories are created (which is a bit of an "insider" way of knowing that linking didn't even run).

**To Reproduce**
`meadow app deploy .` from a Meadow App directory

**Expected behavior**
ILLinker phase should run - at the very least, an error message explaining why it failed. Linker not working is so dangerous OOM wise, that maybe it should not be an optional step.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Developer tools (please complete the following information as best as you can):**
- OS and version: macOS 1.4.3.1
- Meadow CLI: 2.0.41

adrianstevens, 2024-05-16:
Ah so this is expected behavior 

You need to call `meadow app trim` first

adrianstevens, 2024-05-17:
closing issue as this is expected behavior - please feel free to re-open if we want to talk through the CLI behavior

closed 2024-05-17 by adrianstevens