WildernessLabs/Meadow_Issues #639: CLI 2.0.41 fails with startIndex cannot be larger than length of string. (Parameter 'startIndex')

CartBlanche, 2024-05-16:
**Describe the bug**
Attempted `meadow app deploy` on Mac M2 (though customer Gabriel Mongeon is hitting the same error on Windows 11)
after it deletes various files it stops with 
`startIndex cannot be larger than length of string. (Parameter 'startIndex')`

**To Reproduce**
As per above attempt a `meadow app deploy`

**Expected behavior**
the app deploy is successful

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Apple M2 - macOS: 14.4.1 (23E224
 - IDE and version: N/A
 - Meadow extension for IDE version:N/A

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information 
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information 
    Processor type: STM32F777IIK6
    ID: 26-00-25-00-0A-50-30-52-54-31-35-20
    Serial number: 205A31845230
    Coprocessor type: ESP32
    MAC Address - 
	WiFi: 4C:75:25:D6:C7:A8

Firmware Versions 
    OS: 1.11.0.0
    Runtime: 1.11.0.0
    Coprocessor: 1.11.0.0
    Protocol: 8

adrianstevens, 2024-05-16:
The variable `startIndex` only appears in the macOS code path .... I'll take a closer look but this error doesn't make any sense when running on Windows.

CartBlanche, 2024-05-16:
@adrianstevens I'm just going off what the customer told me and what was on the public support post. Maybe ping the customer to confirm?
I only reproed the `startIndex` issue on my mac and it crashed in GetRelativePath path each time, when I stepped through the code.

adrianstevens, 2024-05-16:
I'm running
`meadow app build`
`meadow app trim`
`meadow app deploy`

Using CLI v2.0.41 and it's working correctly.

adrianstevens, 2024-05-17:
There was a bug in the `RelativePath` method - here's a fix along with recursive folder support and some other cleanup
https://github.com/WildernessLabs/Meadow.CLI/pull/565

People can try CLI v2.0.42.0-beta if they want to test. Otherwise expect a v2.0.43 release soon.

adrianstevens, 2024-05-22:
Fixed in CLI v2.0.43

closed 2024-05-22 by adrianstevens