WildernessLabs/Meadow_Issues #641: F7 WiFi network API does not disconnect properly

ctacke, 2024-05-20:
I have an app that tries to disconnect the WiFi network on F7 CCM hardware (Clima, specifically).  I have code that is simple:

```
if (networkAdapter is IWiFiNetworkAdapter wifi)
{
    Resolver.Log.Info("Disconnecting network...");
    await wifi.Disconnect(true);
    Resolver.Log.Info("Network disconnected");
}
```

I see the first log output:
```
Disconnecting network...
```
But the call to `Disconnect()` only ever throws a `TimeoutException`

NevynUK, 2024-06-24:
Three components to this:

- [ ] ESP32 events needed some work (this has been completed and merged to `develop`.  Assigned to @NevynUK 
- [ ] Cell needs to look at the new eventing model and implement the features that are relevant to cell. Assigned to @halyssonJr 
- [ ] Core needs updating the implement the state changes. Assigned @alexischr

alexischr, 2024-07-09:
We are seeing issues with many of our network events succesfully being surface to Meadow.Core, and getting lost somewhere in `Source/implementations/f7/Meadow.F7/Devices/Esp32Coprocessor/Esp32WiFiAdapter.cs`'s `InvokeEvent`.

bryancostanich, 2024-08-06:
Possibly no longer an issue. In @halyssonJr's hands for confirmation.

halyssonJr, 2024-08-12:
@bryancostanich  @ctacke, To confirm, this issue was fixed.

Output
```
Meadow StdOut: [core] Device is configured to use WiFi for the network interface
Meadow StdOut: [core] All cloud features are disabled.
Meadow StdOut: Initialize...
Meadow StdOut: Run...
Meadow StdOut: wifi.IsConnected = False
Meadow StdOut: IPAddress = 255.255.255.255
Meadow StdOut: Connecting to WiFi Network
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: Waiting for wifi to connect loop
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: IPAddress = 255.255.255.255
Meadow StdOut: Waiting for wifi to connect loop
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: IPAddress = 255.255.255.255
Meadow StdOut: Waiting for wifi to connect loop
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: IPAddress = 255.255.255.255
Meadow StdOut: Waiting for wifi to connect loop ended
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: IPAddress = 192.168.15.11
Meadow StdOut: Wifi_NetworkConnected event fired...
Meadow StdOut: wifi.IsConnected = True
Meadow StdOut: Execute wifi.Disconnect now
Meadow StdOut: wifi.IsConnected (from Wifi_NetworkDisconnected event) = False
Meadow StdOut: Reason - ManualDisconnect
Meadow StdOut: wifi.IsConnected = False
Meadow StdOut: IPAddress = 255.255.255.255
```
[WiFiDisconnected.zip](https://github.com/user-attachments/files/16588574/WiFiDisconnected.zip)

bryancostanich, 2024-08-12:
Yay!

closed 2024-08-12 by bryancostanich