WildernessLabs/Meadow_Issues #661: MicroLayout - DisplayLabel should clip text if its outside of width and heigth values

jorgedevs, 2024-05-21:
On a Project Lab v3, conside the following MicroLayoutSample

```csharp
_screen.Controls.Add(
                new DisplayBox(0, 0, _screen.Width, _screen.Height)
                {
                    ForeColor = Color.White
                },
                new DisplayLabel(60, 20, 100, 40)
                {
                    Text = "Welcome to MicroLayout!",
                    TextColor = Color.Black,
                    BackColor = Color.Red,
                    Font = new Font12x20(),
                    VerticalAlignment = VerticalAlignment.Center,
                    HorizontalAlignment = HorizontalAlignment.Center
                });
```

Notice that the background color (red) delimits the bounds of this control, yet on the Text is drawn outside these boundaries. 

[image]

adrianstevens, 2025-04-13:
I've spent some time on this issue and I personally think it's out-of-scope for MicroLayout.

And I recommend this gets reclassified as a feature request.

adrianstevens, 2025-04-13:
Won't fix - closing issue

closed 2025-04-13 by adrianstevens