WildernessLabs/Meadow_Issues #666: Cloud request failing with BadRequest keeps on retrying

jbeenes, 2024-05-22:
The following occured on Linux, not sure if the F7 will have same behaviour. Using version 1.11 nugets.

After a device crash the program wants to log the exception that has occured. 
In my case that failed: "cloud request to /api/logs failed with BadRequest: 'Invalid JSON body'. "

Then the program seems to retry every second or so to make this cloud request and fails ofcourse every time.

This also ruins my logfile :(

duduita, 2024-05-22:
There's a recently merged [PR](https://github.com/WildernessLabs/Meadow.Foundation/pull/985) that should fix it in the upcoming release, since it was a problem on the `Resolver.JsonSerializer.Serialize` which is used by F7 and Linux as well.

jbeenes, 2024-05-22:
Great, will wait for that. Thanks!

closed 2024-05-28 by adrianstevens