WildernessLabs/Meadow_Issues #667: Sleep/Wake is failing -appears to maybe be restarting Mono?

ctacke, 2024-05-22:
I have a simple app doing time sleep/wake about every 8 seconds.  In less than 70 iterations, the device will stop the cycle due to a crash.

The top-level crash is due to a failure to write to the Console.  This is an untrappable exception that causes the device app to just stop because Mono has crashed.  The device does not auto-reset.
```
stderr> [ERROR] FATAL UNHANDLED EXCEPTION: System.IO.IOException: Write
stderr> fault on path /[Unknown]
stderr>   at System.IO.FileStream.WriteInternal (System.Byte[] src, Syst
stderr> em.Int32 offset, System.Int32 count) [0x00077] in <77563d7372d94
stderr> 8198f425b41a9b84c90>:0
stderr>   at System.IO.FileStream.Write (System.Byte[] array, System.Int
stderr> 32 offset, System.Int32 count) [0x00090] in <77563d7372d948198f4
stderr> 25b41a9b84c90>:0
stderr>   at System.IO.StreamWriter.Flush (System.Boolean flushStream, S
stderr> ystem.Boolean flushEncoder) [0x00080] in <77563d7372d948198f425b
stderr> 41a9b84c90>:0
stderr>   at System.IO.StreamWriter.WriteLine (System.String value) [0x0
stderr> 000d] in <77563d7372d948198f425b41a9b84c90>:0
stderr>   at System.IO.TextWriter+SyncTextWriter.WriteLine (System.Strin
stderr> g value) [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
stderr>   at (wrapper synchronized) System.IO.TextWriter+SyncTextWriter.
stderr> WriteLine(string)
stderr>   at System.Console.WriteLine (System.String value) [0x00000] in
stderr>  <77563d7372d948198f425b41a9b84c90>:0
stderr>   at Meadow.Logging.ConsoleLogProvider.Log (Meadow.Logging.LogLe
stderr> vel level, System.String message, System.String messageGroup) [0
stderr> x000b4] in <47e66529e6924cf3bbe5395f260d3c71>:0
stderr>   at Meadow.Logging.Logger.Log (Meadow.Logging.LogLevel level, S
stderr> ystem.String message, System.String messageGroup) [0x000e6] in <
stderr> 47e66529e6924cf3bbe5395f260d3c71>:0
stderr>   at Meadow.Logging.Logger.Error (System.String message, System.
stderr> String messageGroup) [0x00001] in <47e66529e6924cf3bbe5395f260d3
stderr> c71>:0
stderr>   at Meadow.MeadowOS.ReportAppException (System.Exception e, Sys
stderr> tem.String message) [0x00006] in <0c18b10127c64f55b0e612954b587a
stderr> f7>:0
stderr>   at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp ap
stderr> p) [0x002c5] in <0c18b10127c64f55b0e612954b587af7>:0
stderr>   at Meadow.MeadowOS.Main (System.String[] args) [0x0007a] in <0
stderr> c18b10127c64f55b0e612954b587af7>:0
stderr>   at Meadow.MeadowOS.<Main> (System.String[] args) [0x0000c] in
stderr> <0c18b10127c64f55b0e612954b587af7>:0
```
This is happening in `ReportAppException` which indicates things have already gone sideways, and this is not the root cause of failure.
The device is, however, capturing and recording the failure that happened prior to this report, which is:
```
System.TypeLoadException: Could not load type of field 'SleepWakeTest.MeadowApp:statusLabel' (2) due to: Could not resolve type with token 01000010 from typeref (expected class 'Meadow.Foundation.Graphics.MicroLayout.Label' in assembly 'MicroLayout, Version=1.11.0.0, Culture=neutral, PublicKeyToken=null') assembly:MicroLayout, Version=1.11.0.0, Culture=neutral, PublicKeyToken=null type:Meadow.Foundation.Graphics.MicroLayout.Label member:(null)
  at (wrapper managed-to-native) System.RuntimeTypeHandle.type_is_assignable_from(System.Type,System.Type)
  at System.RuntimeTypeHandle.CanCastTo (System.RuntimeType type, System.RuntimeType target) [0x00000] in <77563d7372d948198f425b41a9b84c90>:0
  at System.RuntimeType.IsAssignableFrom (System.Type c) [0x00020] in <77563d7372d948198f425b41a9b84c90>:0
  at Meadow.MeadowOS.FindAppType (System.String root) [0x000a5] in <0c18b10127c64f55b0e612954b587af7>:0
  at Meadow.MeadowOS.FindAppForPlatform (Meadow.Hardware.MeadowPlatform platform) [0x00001] in <0c18b10127c64f55b0e612954b587af7>:0
  at Meadow.MeadowOS.Initialize (System.String[] args, Meadow.IApp app) [0x00093] in <0c18b10127c64f55b0e612954b587af7>:0
  at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00040] in <0c18b10127c64f55b0e612954b587af7>:0
```
This happens very early in the app - before `Run` so it feels like the app has somehow re-entered Main?  This is where it's failing:
https://github.com/WildernessLabs/Meadow.Core/blob/9f40ce0e6f8e20eea8b64715f9eaf6488489e2d5/source/Meadow.Core/MeadowOS.cs#L315

ctacke, 2024-05-22:
Test app is attached.  Generally will fail at ~30 iterations
[TimedSleepTest.zip](https://github.com/WildernessLabs/Meadow_Issues/files/15404931/TimedSleepTest.zip)

alexischr, 2024-06-25:
Bug reproed, needs more investigation

adrianstevens, 2024-07-23:
Also related to #423 
https://github.com/WildernessLabs/Meadow_Issues/issues/423

bryancostanich, 2024-08-27:
Peter is investigating. Some weird thing happening with the interrupt that the RTC is sending. Either the interrupt isn't getting sent, or it's getting lost.

bryancostanich, 2024-09-10:
Looks like an internal alarm misconfiguration. Likely fix in place and being validated internally by Peter. 🎉

bryancostanich, 2024-09-24:
Fixed, to be released in v1.15. There were two internal clocks that had a disagreement between what should be canonical, either 12hrs or 24hrs.

alexischr, 2024-10-29:
Fix released with 1.15

closed 2024-10-29 by alexischr