WildernessLabs/Meadow_Issues #680: Windows - FT232H driver - cannot open I2C bus

houstonhaynes, 2024-05-28:
**Describe the bug**
Getting "App err ijn App Initialization: Error writing device while setting up address" when opening an I2C but on the FT232H (to connect to a PCA9685)

```bash
Creating 'MeadowApp' instance took 16ms
All cloud features are disabled.
Initializing App
Creating Outputs
App Error in App Initialize: Error writing device while setting up address
 System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.FtdiExpander.I2CBus.Write(Byte peripheralAddress, Span`1 writeBuffer, Boolean terminatingStop) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.Ftxxxx\Driver\FtdiExpander.I2CBus.cs:line 85
   at Meadow.Foundation.ICs.IOExpanders.FtdiExpander.I2CBus.Write(Byte peripheralAddress, Span`1 writeBuffer) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.Ftxxxx\Driver\FtdiExpander.I2CBus.cs:line 76
   at Meadow.Foundation.ICs.IOExpanders.Pca9685.PwmPort.Write(Byte register, Byte ledXOnL, Byte ledXOnH, Byte ledXOffL, Byte ledXOffH) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.PwmPort.cs:line 125
   at Meadow.Foundation.ICs.IOExpanders.Pca9685.PwmPort.SetPwm(Byte port, Int32 on, Int32 off) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.PwmPort.cs:line 120
   at Meadow.Foundation.ICs.IOExpanders.Pca9685.PwmPort.Start() in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.PwmPort.cs:line 106
   at Meadow.Foundation.ICs.IOExpanders.Pca9685.PwmPort.set_DutyCycle(Single value) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.PwmPort.cs:line 60
   at Meadow.Foundation.Servos.ServoBase..ctor(IPwmPort pwmPort, ServoConfig config) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\Servos.ServoCore\Driver\ServoBase.cs:line 31
   at Meadow.Foundation.Servos.AngularServoBase..ctor(IPwmPort pwmPort, ServoConfig config) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\Servos.ServoCore\Driver\AngularServoBase.cs:line 24
   at Meadow.Foundation.Servos.Servo..ctor(IPwmPort pwmPort, ServoConfig config) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\Servos.ServoCore\Driver\ServoCore.cs:line 25
   at MeadowApp.MeadowApp.Initialize() in D:\repos\Meadow.Desktop.FSharp\MeadowApp.fs:line 80
   at Meadow.MeadowOS.Start(String[] args, IApp app) in D:\repos\Meadow.Core\source\Meadow.Core\MeadowOS.cs:line 119
App shutting down
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect FT232H to PC
2. Connect SDA/SLC pins from PCA9685 to D0/D1 on FT232H (or pins of choice)
3. Set up IPwmPort that uses the PCA9685 and attempts to use the I2C bus
4. See error above

**Expected behavior**
PWM port should be available for use by the MeadowApp

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [e.g., Windows 11 23H2]
 - IDE and version: [e.g., JetBrains Rider 2024.1.2]
 - Meadow extension for IDE version: N/A

**Meadow (please complete the following information as best as you can):**
N/A

[image]

houstonhaynes, 2024-06-23:
Update with the latest develop branch...

```fsharp
    override this.Initialize() =
        Console.WriteLine("Creating Outputs")
        retractRelay <- Relay(expander.Pins.C3)
        stopRelay <- Relay(expander.Pins.C4)
        extendRelay <- Relay(expander.Pins.C5)
        rainSensor <- expander.Pins.C6.CreateDigitalInputPort(ResistorMode.ExternalPullDown)
        i2CBus <- expander.CreateI2cBus()
        
        pca9685 <- new Pca9685(i2CBus, Frequency(50, Frequency.UnitType.Hertz), (byte)Pca9685.Addresses.Default);
        port0 <- pca9685.CreatePwmPort(pca9685.Pins.LED0, 0.05f)
        wiperServo <- new Sg90(port0)

        Task.CompletedTask
```

```bash
App Error in App Initialize: Error writing device while setting up address
 System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.FtdiExpander.I2CBus.Write(Byte peripheralAddress, Span`1 writeBuffer, Boolean terminatingStop) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.Ftxxxx\Driver\FtdiExpander.I2CBus.cs:line 188
   at Meadow.Foundation.ICs.IOExpanders.FtdiExpander.I2CBus.Write(Byte peripheralAddress, Span`1 writeBuffer) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.Ftxxxx\Driver\FtdiExpander.I2CBus.cs:line 173
   at Meadow.Foundation.ICs.IOExpanders.Pca9685.Initialize() in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.cs:line 88
   at Meadow.Foundation.ICs.IOExpanders.Pca9685..ctor(II2cBus i2cBus, Frequency frequency, Byte address) in D:\repos\Meadow.Foundation\Source\Meadow.Foundation.Peripherals\ICs.IOExpanders.PCA9685\Driver\Pca9685.cs:line 70
   at MeadowApp.MeadowApp.Initialize() in D:\repos\Meadow.Desktop.FSharp\MeadowApp.fs:line 87
   at Meadow.MeadowOS.Start(String[] args, IApp app) in D:\repos\Meadow.Core\source\Meadow.Core\MeadowOS.cs:line 121
App shutting down
Shutdown
```

ctacke, 2024-10-15:
I completely rewrote the I2C side last week.  Tested with multiple devices here, and a separate customer confirmed it worked for him.  I'll close this now, but if you see it again, feel free to re-open or file a new issue

closed 2024-10-15 by ctacke