WildernessLabs/Meadow_Issues #691: F7FeatherV2 stops responding when already powered via 5V then unplug USB

michaelkthompson, 2024-05-31:
Normal operation on our equipment has the Feathers powered via 5V pin. Occasionally I will need connect via USB to monitor logs or perform an update. When I'm done, I'll unplug the USB connection. The Feather then stops responding to Maple and sensor input commands, but does not reboot. We then must reboot to resume normal operations.

adrianstevens, 2024-06-11:
Thanks for reporting @michaelkthompson - this will be priority to fix in the near future

Peter-Moody, 2024-06-14:
Using blinky I was unable to duplicate this issue. Powered F7FeatherV2.d using bench power supply and USB from docking station. Set the bench supply to 5.00vdc and 4.70vdc. In both cases, blinky just kept blinking.

Peter-Moody, 2024-06-14:
Using blinky I was unable to duplicate this issue. Powered F7FeatherV2.d using bench power supply and USB from docking station. Set the bench supply to 5.00vdc and 4.70vdc. In both cases, blinky just kept blinking.
Perhaps the customer is drawing more current that my test setup.

Peter-Moody, 2024-06-15:
Load test. 
This test will increasing the current load on the 3v3 power line to determine if this may allow duplication of the issue. The test used 5 - 22 ohm resistors from GND to 3v3. Each resistor will increase the current load by about 150 ma (I=E/R, 3.3v/22r = 0.15A). Due to the breadboard style of the test hookup some voltage drop is expected and therefore the 150 ma is approximate (e.g. 3.0/22 = 0.136).
Results
With 5 - 22 ohm resistors as described, the F7Feather's behavior did not changes. USB power could be disconnected and reconnected without blinky stopping. The voltage on the bench power supply was adjusted to 5.2v, 5.0v and 4.8v with no change in behavior. The only difference was the different voltages the current draw from the bench power supply would vary. This was an expected behavior. At 4.8v the current draw was 0.0 ma because all of the current was coming from the USB connection.

Peter-Moody, 2024-06-15:
I wondered if the 5 volt supply might not be able to deliver the current needed the instant the USB connector was unplugged. So, replaced the bench power supply with 3 different "wall warts" that output via an USB connector 5 volts. None of these 3 failed to provide power when the F7Feather USB cable was disconnected.
At this point I have not been able to duplicate this problem.

Peter-Moody, 2024-06-24:
Thanks to suggestion by @ctacke found the issue. If HCOM is sending a message (via USB) to a host PC and some app is consuming this data (e.g. `meadow listen`) and the USB cable is unplugged while sending, but +5v is still being supplied to Meadow, then the write call within HCOM will return an ENOTCONN (128) error. This error is not expected by the caller, which is reading `stdout`.

Modified HCOM to trap ENOTCONN and return properly under this condition.

This was tested with a modified version of `Blinky` that sends text on every color change (also delay changed from 1000 to 1) and using `meadow listen` for the receiving app.

closed 2024-10-11 by alexischr

NevynUK, 2024-11-18:
Marking as Done as the original issue is now closed..